torbennehmer/hacs-e3dc #8: Add energy sensors

goebelmeier, 2023-06-05:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Hi Torben,
Congratulations on the cool first release. I am relatively close to replacing my own add-on, which is much more complex and heavyweight, with your native HA integration. What I would still miss, besides all the current power values are the energy values to use them inside the HA energy dashboard.

### Describe the solution you'd like

Therefor we need at least the following sensors:
- Solar Energy, based on `TAG_DB_HISTORY_DATA_DAY` `TAG_DB_DC_POWER` (`0x06800004`)
- Battery Energy Charge, based on `TAG_DB_HISTORY_DATA_DAY` `TAG_DB_BAT_POWER_IN` (`0x06800002`)
- Battery Energy Discharge, based on `TAG_DB_HISTORY_DATA_DAY` `TAG_DB_BAT_POWER_OUT` (`0x06800003`)
optional:
- Wallbox Energy (don't know the RSCP tag yet)

All sensors need the following attributes:
- `unit_of_measurement: "kWh"`
- `device_class: "energy"`
- `state_class: total_increasing`

### Describe alternatives you've considered

./.

### Additional context

./.

torbennehmer, 2023-06-05:
@goebelmeier I can do that, should not be much of a problem.

However, please be aware, that those counters have a certain variance. With my S10 I'm usually unable to really match them with the energy meter in my main fuse box and with the energy meter built into the E3DC. I haven't really found out what's going wrong in the E3DC's value, but differences up to 10 to 20 percent in the values to/from grid are quite common here. Maybe this is different with other E3DC systems, can't tell for sure.

torbennehmer, 2023-06-05:
@goebelmeier I have just toyed around with those setting a bit: So far as I can tell my S10 does need a timezone adaption to work correctly. Somehow this seems broken though.

If I send him todays midnight timestamp in UTC, I get data computed starting yesterday, 22:00 local time (CEST+2). So it seems that the clock of my E3DC runs in local time, to correctly get today's data, I've to ask the S10 for 02:00 UTC = 00:00 CEST+2.

Can you confirm that with your unit and what type of unit do you use?

goebelmeier, 2023-06-05:
I do have an S10 E from 2019 running S10_2023_02 with 6,5 kWh LG battery. If you have some sample code to use, i can try to reproduce that issue. Currently i'm running rscp2mqtt and i do see this phenomenon as i do have much too high energy values when restarting my add-on in the middle of the day.

AndreasTausch, 2023-06-06:
Thanks first of all for the great integration.
Unfortunately, except for the "Installed Battery Capacity", no other S10 sensors are displayed in the selection fields of the Energy Dashboard, although the entities are all present and also provide correct values.

So seems I have the same ambition to see the values in the energy dashboard. 

Please let me have a short info what I might change in the sensor configuration.

torbennehmer, 2023-06-08:
@AndreasTausch: I'm working on an update to provide appropriate sensors using the history data of the E3DC but due to some timezone weirdness, this might take a moment to get it running.

Until then, you can use the [Riemann sum integral](https://www.home-assistant.io/integrations/integration/) integration to create a running total over the current sensors.

torbennehmer, 2023-06-08:
> I do have an S10 E from 2019 running S10_2023_02 with 6,5 kWh LG battery. If you have some sample code to use, i can try to reproduce that issue. Currently i'm running rscp2mqtt and i do see this phenomenon as i do have much too high energy values when restarting my add-on in the middle of the day.

If you use the pye3dc module, you can try something like this:

```python
from e3dc import E3DC
import json

TCP_IP = '10.128.20.15'
USERNAME = 'torben@nehmer.net'
PASS = '...'
KEY = '...'
CONFIG = {}

e3dc = E3DC(E3DC.CONNECT_LOCAL, username=USERNAME, password=PASS, ipAddress=TCP_IP, key=KEY)

print(json.dumps(e3dc.get_db_data(), indent=2, default=str))
print(json.dumps(e3dc.get_db_data_timestamp(startTimestamp=<<insert start timestamp here>>,
    timespanSeconds=86400), indent=2, default=str))
print(json.dumps(e3dc.sendRequestTag("INFO_REQ_TIME_ZONE"), indent=2, default=str))
print(json.dumps(e3dc.sendRequestTag("INFO_REQ_UTC_TIME"), indent=2, default=str))
print(json.dumps(e3dc.sendRequestTag("INFO_REQ_TIME"), indent=2, default=str))
```

In the get_db_data_timestamp call, use the timestamp of midnight modified by your current timezone (e.g. +7.200 seconds for CEST) and then subtract 15 minutes = -900 seconds (weirdness in the e3dc portal), that should get you the same values as the portal for the day in question to validate things.

The other lines give the current timestamps for comparison.

torbennehmer, 2023-06-09:
@goebelmeier: I've boosted my testing code a bit, could you try this one and give me the output:

Change the line `tz_local=pytz.timezone("Europe/Berlin")` to your local timezone if you're outside of CE(S)T, let me know what TZ you use instead. Please add the current local time your system's showing when you execute the script.

```python
from e3dc import E3DC
import json
import pytz
from datetime import datetime, date, time

TCP_IP = '10.128.20.15'
USERNAME = 'xxx'
PASS = 'xxx'
KEY = 'xxx'
CONFIG = {}

e3dc = E3DC(E3DC.CONNECT_LOCAL, username=USERNAME, password=PASS, ipAddress=TCP_IP, key=KEY)

e3dc_tz=e3dc.sendRequestTag("INFO_REQ_TIME_ZONE")
e3dc_local=e3dc.sendRequestTag("INFO_REQ_TIME")
e3dc_local_dt=datetime.fromtimestamp(e3dc_local)
e3dc_utc=e3dc.sendRequestTag("INFO_REQ_UTC_TIME")
e3dc_utc_dt=datetime.fromtimestamp(e3dc_utc)
delta=e3dc_local - e3dc_utc

print(f"E3DC Timezone: {e3dc_tz}, delta based on timestamps: {delta}")
print(f"e3dc-Local: {e3dc_local} = {e3dc_local_dt}")
print(f"e3dc-UTC: {e3dc_utc} = {e3dc_utc_dt}")

# E3DC timezone datetime
# May raise Unknown Timezone Error
aware_e3dc = datetime.now(pytz.timezone(e3dc_tz))
print('E3DC-TZ based local DateTime', aware_e3dc)

tz_local=pytz.timezone("Europe/Berlin")
print("tz_local", tz_local)

today = datetime.now(tz_local).replace(hour=0, minute=0, second=0, microsecond=0)

print ("Today", today)
print ("Today tuple", today.timetuple())

today_ts = int(today.timestamp())
print ("Today_ts", today_ts)

e3dc_tzinfo = pytz.timezone(e3dc_tz)
tmp = datetime.now(e3dc_tzinfo)
print("e3dc_tzinfo utcoffset", tmp.utcoffset().seconds)

today_ts += tmp.utcoffset().seconds
today_ts -= 900 # Portal Offset
today_ts -= 86400 # go to yesterday

print ("Today_ts patched", today_ts)

print(json.dumps(e3dc.get_db_data_timestamp(startTimestamp=today_ts,
    timespanSeconds=86400), indent=2, default=str))
```

The values you get should match the aggregated data you say for yesteray in the E3DC portal. If you run it against today (just drop `today_ts -= 86400`, you should get current values plus or minus some tolerance, as the portal data lags behind the live data from the unit.

Be vary, I'm not yet sure if grid_power_in/out are labeld correctly, could be that they're swapped. If they are, it would be great if you let me know that as well.

Sample:

```
E3DC Timezone: Europe/Berlin, delta based on timestamps: 7199.999979972839
e3dc-Local: 1686321747.000402 = 2023-06-09 16:42:27.000402
e3dc-UTC: 1686314547.000422 = 2023-06-09 14:42:27.000422
E3DC-TZ based local DateTime 2023-06-09 14:42:25.869099+02:00
tz_local Europe/Berlin
Today 2023-06-09 00:00:00+02:00
Today tuple time.struct_time(tm_year=2023, tm_mon=6, tm_mday=9, tm_hour=0, tm_min=0, tm_sec=0, tm_wday=4, tm_yday=160, tm_isdst=1)
Today_ts 1686261600
e3dc_tzinfo utcoffset 7200
Today_ts patched 1686181500
{
  "autarky": 79.96076965332031,
  "bat_power_in": 4340.0,
  "bat_power_out": 4617.0,
  "consumed_production": 96.46310424804688,
  "consumption": 29572.0,
  "grid_power_in": 867.0,
  "grid_power_out": 5926.0,
  "startTimestamp": 1686181500,
  "stateOfCharge": 7.771999835968018,
  "solarProduction": 27272.0,
  "timespanSeconds": 86400
}
```

torbennehmer, 2023-06-14:
@goebelmeier Thanks for the test run, that matches what I'm seeing. I'm currently building some first implementation, stay tuned.

bf8392, 2023-07-09:
> @AndreasTausch: I'm working on an update to provide appropriate sensors using the history data of the E3DC but due to some timezone weirdness, this might take a moment to get it running.
>
> Until [...]

Can you show how to configure the Riemann helper? Maybe it would be enough to mention this in docs...because I think it gives nice mulitple values this way...

torbennehmer, 2023-07-09:
Current Git HEAD has a first implementation of the integrated sensors. I'll get a release out in a few days, as soon as I cleanup a few more things, the sensor names are highly inconsistent right now, so I'd advise to wait for the next release for productive usage.

closed 2023-07-09 by torbennehmer

torbennehmer, 2023-07-09:
@bf8392 As for the Riemann Integration, it looks quite simple, before, with my ModBus integration, it looked something like this:

```yaml
- platform: integration
  unique_id: e3dc_netzbezug_menge
  name: "E3DC: Netzbezug Menge"
  source: sensor.e3dc_netzbezug_leistung
  unit_prefix: k
  round: 2
  method: left

- platform: integration
  unique_id: e3dc_netzeinspeisung_menge
  name: "E3DC: Netzeinspeisung Menge"
  source: sensor.e3dc_netzeinspeisung_leistung
  unit_prefix: k
  round: 2
  method: left

- platform: integration
  unique_id: e3dc_pv_menge
  name: "E3DC: PV Menge"
  source: sensor.e3dc_pv_leistung
  unit_prefix: k
  round: 2
  method: left
```

bf8392, 2023-07-09:
Wow great great thanks! That's absolutely Amazing!!! :-)

torbennehmer, 2023-07-09:
@goebelmeier @bf8392 Just one more info: From my point of view, the E3DC DB values are with a significant tolerance, I've seen 10% more than once when comparing to my Tibber Smart Meter reader. I am currently not sure who or what is correct here, so this will stay a bit of a guessing game for now. I'll also have to compare this to the additional energy meter built into my E3DC I use for my taxes. I'll keep an eye on it, but I can't tell how much time it'll take to really get an handle on this.

torbennehmer, 2023-07-09:
Here you go, v2.0.0 will help you:

[image]

Please test, I'd appreciate any feedback. If you experience problems, please open a new bug.

NMozer, 2024-02-02:
Hello,
Sorry i have to ask this.
How can i get the aktuall values to show in the Energy Dashboard? To see the current consumption values.
They need to be "device_class: ENERGY not POWER" to be selectabel in the Dashboard. 
Why do i need the Todays Values in the dashboard? 

SensorEntityDescription(
        key="solar-production",
        translation_key="solar-production",
        icon="mdi:solar-power",
        native_unit_of_measurement=UnitOfPower.WATT,
        suggested_unit_of_measurement=UnitOfPower.KILO_WATT,
        suggested_display_precision=2,
        device_class=SensorDeviceClass.**POWER**,
        state_class=SensorStateClass.MEASUREMENT,
T
can somebody please explain to me. 
Thanks

AndreasTausch, 2024-05-31:
> [Verfolgungsbild][https://github.com/notifications/beacon/A7Q2LEN7BSJPCKTPMROXPDLXLIAH7A5CNFSM6AAAAAAY2QFMDKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTS64DPTI.gif]

Dear Torben,
Still using the version 1.0.0. of your great Integration. Was always Ranking bery well.

What about the sequence to update? Only download the newest Release via HACS and the reboot?

Or better deinstall, Download and New Installation?

Thanks for your advice!

Andreas

PR #16 (2023-07-09): Add energy sensors (for dashboard)
- Add todays energy consumption as individual sensors to HA.
- This enables full integration into the energy dashboard.
- Currenly, there's an heuristic that will compensate for E3DCs skew of the Unix Timestamp value and for the presumed 15 minutes offset during midnight. This will need testing, as the exact behaviour is not yet completely validated. As a worst case, your energy consumption may be 15 minutes off.
- Be aware, that the historical information is only pulled once per minute, so you might have some minor rounding errors during midnight when moving [... cut]