torbennehmer/hacs-e3dc #15: Sensors do not show in Energy Dashboard

bf8392, 2023-07-09:
### System Health details

none

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

When I try to add the Sensors in HA Energy Dashboard, they do not show up as selectable because of missing statistics: https://developers.home-assistant.io/docs/core/entity/sensor#long-term-statistics

### Reproduction steps

1. Install HACS
2. Install the Package
3. Add E3DC
4. Try to add to Energy Dashboard
...

### Debug logs

```text
none
```

goebelmeier, 2023-07-09:
Duplicates #8 from my point of view

bf8392, 2023-07-09:
Oh I'm sorry...

closed 2023-07-09 by bf8392