torbennehmer/hacs-e3dc #23: Add additional Power Meters

bullitt186, 2023-07-17:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I have an additional Power Meter (E3DC LM3p80hhc) connected to my E3DC S10X Compact in order to measure the energy consumed by my heat pump. Within the E3DC Portal, it shows as a "externer Verbrauch" as you can see in the images below. 
With the hacs-e3dc integration version 2.0.0, i cannot see this consumer in home assistant as a separate entity. 

[image]
[image]
[image]

### Describe the solution you'd like

I want to be able to see this consumption data as a separate sensor in Home Assistant.
According to [this issue](https://github.com/fsantini/python-e3dc/issues/61) in the python-e3dc project, multiple meters can be queried via get_powermeter_data() / get_powermeters_data().

A good way could be to add additional additional power meter entities (similar to the "House consumption" & "House consumption - today"), named Power Meter 0 ... 7 which are hidden/disabled by default. By that, the majority of the users who very likely have only one powermeter won't be confused.

### Describe alternatives you've considered

I found no other integration (e.g. via Modbus) which covers additional powermeters so far.

### Additional context

My Python / Integrations skills are insufficient to implement the request on my own, unfortunately.

torbennehmer, 2023-07-18:
Hi @bullitt186,
that sounds like a good idea. To get a better understanding about the way your unit reacts, would you mind to update to release 2.1 of the integration and create a diagnostics dump? 
As there is some sensitve information like serials and MAC adresses in it, please upload it at the following location, please use the issue ID somewhere in the filename so that I can distinguish it: https://kurzer.link/e3dcrscp
Cheers, Torben

bullitt186, 2023-07-18:
Hi Torben, I will do so tomorrow. However I explored this evening the options as well. What I can see so far:
python-e3dc shows the additional powermeter as an additional source.

e3dc.poll() via Web Connection reveals this: key "add" within the production:
{'time': datetime.datetime(2023, 7, 18, 20, 8, 49, tzinfo=datetime.timezone.utc), 'sysStatus': '4211391', 'stateOfCharge': 91, 'production': {'solar': 0, **'add': -2884**, 'grid': -22}, 'consumption': {'battery': -3504, 'house': 3482, 'wallbox': 0}}
Oddly enough, this value isn't shown in Home Assistant:
[image]
I couldn't see any obvious error compared to the other signals for this one in your code.

When calling e3dc.poll_ajax_raw(), we see that the portal returns for PM1 to PM4 values, where in my case PM1 is the heat pump power meter. 
{'TSMS': '1689704090000', 'POWER_BAT': '-3699', 'POWER_LM_L1': '338', 'POWER_LM_L2': '-218', 'POWER_LM_L3': '-138', 'POWER_WALLBOX': '0', 'serialno': '812325015689', 'time': '2023-07-18 20:14:50', 'IP': '1311478323', 'dataVersion': '1', 'offlineLevel': '0', 'SYSSTATUS': '4211391', 'POWER_BAT_1': '-1233', 'POWER_BAT_2': '-1233', 'POWER_BAT_3': '-1233', 'POWER_PV_S1': '0', 'POWER_PV_S2': '0', 'POWER_PV_S3': '0', 'POWER_AC_L1': '1200', 'POWER_AC_L2': '1168', 'POWER_AC_L3': '1202', 'POWER_ROOTLM_L1': '338', 'POWER_ROOTLM_L2': '-218', 'POWER_ROOTLM_L3': '-138', 'POWER_C_L1': '1581', 'POWER_C_L2': '993', 'POWER_C_L3': '1107', 'SOC': '88', 'PM_ALIVE': '3', **'PM1_L1': '1285', 'PM1_L2': '842', 'PM1_L3': '878', 'PM2_L1': '0', 'PM2_L2': '0', 'PM2_L3': '0', 'PM3_L1': '0', 'PM3_L2': '0', 'PM3_L3': '0', 'PM4_L1': '0', 'PM4_L2': '0', 'PM4_L3': '0'**, 'WB_ALIVE': '128', 'WB0_L1': '0', 'WB0_L2': '0', 'WB0_L3': '0', 'WB0PV_L1': '0', 'WB0PV_L2': '0', 'WB0PV_L3': '0', 'WB1_L1': '0', 'WB1_L2': '0', 'WB1_L3': '0', 'WB1PV_L1': '0', 'WB1PV_L2': '0', 'WB1PV_L3': '0', 'WB2_L1': '0', 'WB2_L2': '0', 'WB2_L3': '0', 'WB2PV_L1': '0', 'WB2PV_L2': '0', 'WB2PV_L3': '0', 'WB3_L1': '0', 'WB3_L2': '0', 'WB3_L3': '0', 'WB3PV_L1': '0', 'WB3PV_L2': '0', 'WB3PV_L3': '0', 'WB4_L1': '0', 'WB4_L2': '0', 'WB4_L3': '0', 'WB4PV_L1': '0', 'WB4PV_L2': '0', 'WB4PV_L3': '0', 'WB5_L1': '0', 'WB5_L2': '0', 'WB5_L3': '0', 'WB5PV_L1': '0', 'WB5PV_L2': '0', 'WB5PV_L3': '0', 'WB6_L1': '0', 'WB6_L2': '0', 'WB6_L3': '0', 'WB6PV_L1': '0', 'WB6PV_L2': '0', 'WB6PV_L3': '0', 'PM7_L1': '0', 'PM7_L2': '0', 'PM7_L3': '0', 'LM_COUNT': '2'}

So my inital suggestion to add these PMs as hidden entities could work.
What I do not get, is how you can access these values from the hacs integration, as these values are not exposed by python-e3dc (compared to the other ones such as house consumption, etc. does this need a PR to python-e3dc?

torbennehmer, 2023-07-30:
@bullitt186 It would be great if you could send me a diagnostic dump as mentioned before, I'd like to get an impression on how your system looks like when queried in detail.

bullitt186, 2023-07-31:
@torbennehmer I just dropped you a diagnostic dump on the URL above.
I played a couple of days ago with 2.0.0 and added the additional powermeters on my own, see the changes below.
The reason why I didn't create a PR to your project yet is, that get_powermeters_data from python-e3dc crashes when asked to collect powermeter data from nonexisting powermeters. For that reason, i created a PR which isn't merged yet to python-e3dc to fix that: https://github.com/fsantini/python-e3dc/pull/79.
With this Fix, the changes below work well and provide 4 additional powermeters. 
If you find another / better way to include the powermeters without changes to the python-e3dc this would be awesome!

### coordinator.py:

**Configure the Powermeters**
`def create_e3dcinstance(username: str, password: str, host: str, rscpkey: str) -> E3DC:
    CONFIG = {
        "powermeters": [
            {"index": 0},
            {"index": 1},
            {"index": 2},
            {"index": 3},
            {"index": 4},
        ]
    }
    e3dc = E3DC(
        E3DC.CONNECT_LOCAL,
        username=username,
        password=password,
        ipAddress=host,
        key=rscpkey,
        configuration=CONFIG,
    )
    return e3dc`

**Collecting the Powermeter Data:**
`        powermeters_data: dict[str, Any] = await self.hass.async_add_executor_job(
            self.e3dc.get_powermeters_data
        )
        self._process_powermeters_data(powermeters_data)
`

**Processing the Powermeter Data:**
`    def _process_powermeters_data(self, powermeters_data: dict[str, Any]):
        for powermeter in powermeters_data:
            if (
                powermeter["index"] > 0 and powermeter["index"] <= 4
            ):  # The first Powermeter is the House Powermeter and no additional one.
                self._mydata["additional-powermeter" + str(powermeter["index"])] = (
                    int(powermeter["power"]["L1"])
                    + int(powermeter["power"]["L2"])
                    + int(powermeter["power"]["L3"])
                )`

### sensors.py
` # ADDITIONAL POWER METER SENSORS
    SensorEntityDescription(
        key="additional-powermeter1",
        translation_key="additional-powermeter1",
        icon="mdi:meter-electric",
        native_unit_of_measurement=UnitOfPower.WATT,
        suggested_unit_of_measurement=UnitOfPower.KILO_WATT,
        suggested_display_precision=2,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        entity_registry_enabled_default=False,
    ),
    SensorEntityDescription(
        key="additional-powermeter2",
        translation_key="additional-powermeter2",
        icon="mdi:meter-electric",
        native_unit_of_measurement=UnitOfPower.WATT,
        suggested_unit_of_measurement=UnitOfPower.KILO_WATT,
        suggested_display_precision=2,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        entity_registry_enabled_default=False,
    ),
    SensorEntityDescription(
        key="additional-powermeter3",
        translation_key="additional-powermeter3",
        icon="mdi:meter-electric",
        native_unit_of_measurement=UnitOfPower.WATT,
        suggested_unit_of_measurement=UnitOfPower.KILO_WATT,
        suggested_display_precision=2,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        entity_registry_enabled_default=False,
    ),
    SensorEntityDescription(
        key="additional-powermeter4",
        translation_key="additional-powermeter4",
        icon="mdi:meter-electric",
        native_unit_of_measurement=UnitOfPower.WATT,
        suggested_unit_of_measurement=UnitOfPower.KILO_WATT,
        suggested_display_precision=2,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        entity_registry_enabled_default=False,
    ),`

nlkoOlli, 2023-08-10:
Would be nice if the change will be put in the next version. Thanks for your work.

bullitt186, 2023-08-31:
Just a heads up:
We have now completed one prerequiste to implement additional powermeters: ``python-e3dc`` got in Rel. 0.7.5 ``get_powermeters()`` (https://github.com/fsantini/python-e3dc/pull/80), so that the installed powermeters can be identified.

bullitt186, 2023-09-17:
I implemented the additional powermeters in [PR57](https://github.com/torbennehmer/hacs-e3dc/pull/57)

torbennehmer, 2023-09-18:
Thanks for the PR, please give me a bit to look if I can find a way to get the translations in so that we've got better names for the entities. I've to look when I find a bit of time for it. I'll tackle that next after the current release.

bullitt186, 2023-09-18:
after a good night of sleep i had the solution within a couple of minutes, see updated PR. Let's continue the discussion on the PR in case you have more feedback & thoughts.

torbennehmer, 2023-10-13:
@bullitt186 Thanks again for your work, I've just pulled the PR in, I'll fix the Ruff Error myself, so don't worry. Thanks again.

closed 2023-10-13 by torbennehmer

bullitt186, 2023-10-13:
Thank you as well for your patience :-)