torbennehmer/hacs-e3dc #40: Impossible state class

derjoerg, 2023-08-03:
### System Health details

## System Information

version | core-2023.7.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.39
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4986
Installed Version | 1.32.1
Stage | running
Available Repositories | 1336
Downloaded Repositories | 64
HACS Data | ok

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 16. September 2023 um 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | eu-central-1-10.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Discovergy</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.4
-- | --
update_channel | stable
supervisor_version | supervisor-2023.07.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 48.5 GB
disk_used | 27.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (15.0.6), Samba share (10.0.2), Mosquitto broker (6.2.1), Node-RED (14.4.3), Samba Backup (5.2.0), File editor (5.6.0), ioBroker (0.8.0), ESPHome (2023.7.1), Signal Messenger (0.67.1), Matter Server (4.9.0), Music Assistant BETA (2.0.0b67), Speedtest (1.5.0), Piper (0.1.3), Whisper (0.2.1), Studio Code Server (5.10.0), PSA Car Controller (v3.0.12)

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 37
views | 56
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 24. Juli 2023 um 08:55
-- | --
current_recorder_run | 3. August 2023 um 14:10
estimated_db_size | 6658.47 MiB
database_engine | mysql
database_version | 10.11.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I get the following during startup:

```
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:505
Integration: Sensor (documentation, issues)
First occurred: 14:11:02 (1 occurrences)
Last logged: 14:11:02

Entity sensor.s10x_energy_charged_from_grid (<class 'custom_components.e3dc_rscp.sensor.E3DCSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
```

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
see above
```

torbennehmer, 2023-08-04:
@derjoerg This could be a problem with the upgrade. To rule this out, could you do one thing: Delete the integration from HA, fully restart HA (just to be on the safe side) then reinstall the integration?

derjoerg, 2023-08-04:
Hi,
still the same.
Steps done:

1. Deleted the configuration
2. Removed from HACS
3. Restarted HA
4. Added from HACS
5. Restarted HA
6. Configured integration

Same error

torbennehmer, 2023-08-04:
Ok I'll look into it. Probably won't come sound to it till Monday it Tuesday.

torbennehmer, 2023-08-07:
Error is fixed, new release will arrive as soon as I can.

torbennehmer, 2023-08-07:
Actually, there are two problems with this sensor. The wrong state class slipped my mind. When looking deeper into it once more when retesting I discovered, that this counter actually is Ah per battery Cell (which is at around 3,65V in my unit). This definitely counts as a creative way of doing this. For the moment, I'll use the 3,65V as multiplier to get a quick fix and the warning away. However, this will need more work once the battery details have been determined. I assume that the actual voltage per individual Battery cell is not always 3,65.

PR #41 (2023-08-07): Fix manual charged energy sesnsor and work around its kAh weirdness.
- fixes #40 
- will convert the reading to kWh. Seems to be in kAh, so when multiplied by my own 3,65 V we result in the correct amount of kWh.
commit 130ade42 (2023-08-07): Fix manual charged energy sesnsor and work around its kAh weirdness. (#41)

- fixes #40 
- will convert the reading to kWh. Seems to be in kAh, so when
multiplied by my own 3,65 V we result in the correct amount of kWh.
closed 2023-08-07 by torbennehmer