torbennehmer/hacs-e3dc #43: Additional Production

MarcoAVoegele, 2023-08-08:
### System Health details

./.

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

There is no "sensor.s10e_pro_additional_production" or similar showing up in entities, although there is an E3DC external inverter with additonal 2 strings installed.

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
./.
```

torbennehmer, 2023-08-09:
Please add a diagnostics dump. If you don't want to upload it here, you can use https://kurzer.link/e3dcrscp as an alternative.

MarcoAVoegele, 2023-08-09:
Found it… It was disabled… I have enabled it and now it is visible. Sorry – My fault! Great stuff!!!

torbennehmer, 2023-08-09:
@MarcoAVoegele Yes, that's true, those sensors are not active by default. Please drop me a diagnostic dump nevertheless: I'd like to check if I can automatically enable the sensor in some way. To get into this, I'd need that dump, my own unit does not have an additional source, thus I cannot really analyze this no my end.

MarcoAVoegele, 2023-08-09:
Hi, yes I can do it. What do I need to do, so that you get the required information?

torbennehmer, 2023-08-09:
Go to the device in the HA setup, klick the tripledot menu and download the diagnostics dump from there. It'll contain a lot of information queried from the unit, that give me an impression how it reacts to API calls.

[image]

MarcoAVoegele, 2023-08-09:
It does not download the file…
***@***.***

[Das Bild wurde vom Absender entfernt. image]<https://user-images.githubusercontent.com/2812067/259502314-16514e44-0de8-46be-87a2-98d2f41e330b.png>

torbennehmer, 2023-08-10:
Please check if you're using the latest version of the integration.

torbennehmer, 2023-10-20:
Release 3.4 should add all additional productions as their own powermeters. Please validate this against your requirements. If it works, it would be great if you could send me one final diagnostics dump, I'll probably deprecate the old sensors at this point, but I'd like to verify if possible.

derhappy, 2023-11-24:
We had the same issue until I found the disabled entity. However, here's the requested dump.
Besides the E3DC system we have one additional production meter as well as one additional consumption meter
 
[e3dc_dump.json](https://github.com/torbennehmer/hacs-e3dc/files/13462354/e3dc_dump.json)

closed 2026-10-16 by torbennehmer