torbennehmer/hacs-e3dc #44: Add entities for energy production of individual strings

bullitt186, 2023-08-20:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I have Solar Panels on both sides of my roof. one facing to the east, one facing to the west.
In order to get a good forecast of my PV production, i have set up two roofs in the [Forecast.Solar](https://www.home-assistant.io/integrations/forecast_solar) Integration.
However the forecast is not very accurate. I want to tune the parameters of the Forecast per roof and match it with the actual production. Therefore, I'd like to get the output of the individual strings so i can compare the the string output for each roof side with the forecast per roof side.

### Describe the solution you'd like

Today, the Integration only delivers the combined power from the PV modules in ``sensor.s10x_solar_production``.
However the python_e3dc library delivers via ``get_pvi_data()`` a strings object:
```yaml
"strings": { 0:
                        {
                            "power": <power>,
                            "voltage": <voltage>,
                            "current": <current>,
                            "energyAll": <energy all>
                        }
```
I'd like to have the energyAll per String as an entity in the E3DC HA integration.

### Describe alternatives you've considered

I could hack a quick and dirty solution via node red or something similar but I like it neat and tidy and think others may benefit from this feature as well :-)

### Additional context

Similar to the discussions on additional powermeters [here](https://github.com/fsantini/python-e3dc/pull/79), ideally there would be a way to identify the installed strings so that the entities can be created during configuration of the integration.

kwitt21, 2023-10-25:
I would also really appreciate the ability to display both strings/trackers separately in Home Assistant!

TommiG1, 2026-10-23:
Covered by #407.

Lifetime energy per string is included as `pvi-string-N-energy`. Live tested on an S10E Pro.