torbennehmer/hacs-e3dc #46: Download of diagnostic information for hacs-e3dc not possible (500 internal server error)

neuromancer3142, 2023-09-01:
### System Health details

## System Information

version | core-2023.8.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1273
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.08.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 30.8 GB
disk_used | 7.5 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | InfluxDB (4.7.0), Grafana (9.0.3), File editor (5.6.0), Terminal & SSH (9.7.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 9
views | 8
mode | storage

</details>

<details><summary>MercedesME 2020</summary>

websocket_connection_state | connected
-- | --
api_endpoint_reachable | ok

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25. August 2023 um 06:06
-- | --
current_recorder_run | 1. September 2023 um 13:21
estimated_db_size | 405.98 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I apologise for any mistakes upfront, as this is my *first* time I post an issue or perform any activity on GitHub. So happy for any feedback on how to improve and get better.

Encountered issue: Download of diagnostic information for hacs-e3dc is possible, but the downloaded html file shows:

"500 internal server error
Server got itself in trouble"

I performed the steps as advised by Torben:

"Go to the device in the HA setup, klick the tripledot menu and download the diagnostics dump from there. It'll contain a lot of information queried from the unit, that give me an impression how it reacts to API calls."

Seems that I have the same problem as described in:
https://github.com/torbennehmer/hacs-e3dc/issues/43

### Reproduction steps

1. Go to the device in the HA setup, klick the tripledot menu and download the diagnostics dump from there. It'll contain a lot of information queried from the unit, that give me an impression how it reacts to API calls.
2. Downloaded file will always show the same error message as indicated above.
3.
...

### Debug logs

```text
The error message I get in /config/home-assistant.log:

*** start ***
TypeError: unsupported format string passed to NoneType.__format__
2023-09-01 10:34:40.684 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 148, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/diagnostics/__init__.py", line 249, in get
    data = await info.config_entry_diagnostics(hass, config_entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 32, in async_get_config_entry_diagnostics
    "get_powermeters_data": e3dc.get_powermeters_data(),
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 1853, in get_powermeters_data
    self.get_powermeter_data(
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 1810, in get_powermeter_data
    activePhases = f"{activePhasesChar:03b}"
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: unsupported format string passed to NoneType.__format__
*** stop ***

A full debug log is also available, but has 380 MB ... Please let me know if you need it.
```

torbennehmer, 2023-09-03:
@neuromancer3142 Thanks for bringing up your various issues and sorry for not answering faster, I've been on vacation during most of August, so a few things piled up. 

The error you indicated came from pye3dc below, however, adding some resiliency to diagnostics.py is clearly in order.

Current Git HEAD has been bumped to 0.7.6, can you try if this fixes your problem along the way so we get a debug dump again?

neuromancer3142, 2023-09-03:
Thanks @torbennehmer for your quick response. Perhaps this is a very silly question: Is there a new release I can install? Or do I need to reinstall my current release in HACS in order to get this updated version? I only see the 3.2.0 release from last month here on GitHub or is a new release imminent?

torbennehmer, 2023-09-12:
@neuromancer3142 You'd have to deploy the module manually by cloning git / downloading the codebase from Github.
Anyway, I've had more problems like these reported in the last week, so I'm going to harden the diagnostics function to start with, so that it fails more gracefully and then release again. That should bring us forward and make it easier for you to test things.

Apart from that, the actual error there seems quite stange to me, it is coming directly from the pye3dc base library. I am wondering what fails there, maybe your unit is dropping powermeter data in a strange way. You might want to try this Python script on the HA server's command line, that should reproduce the issue:

```python
from e3dc import E3DC
import json

TCP_IP = '10.128.20.10'
USERNAME = '...'
PASS = '...'
KEY = '...'
CONFIG = {}

e3dc = E3DC(E3DC.CONNECT_LOCAL, username=USERNAME, password=PASS, ipAddress=TCP_IP, key=KEY)
print(json.dumps(e3dc.get_powermeters_data(), indent=2, default=str))
```

Expected behavoir would be an prettyprinted response from the library, similar to this one:

```json
[
  {
    "activePhases": "111",
    "energy": {
      "L1": -4406586.59,
      "L2": 18933825.74,
      "L3": 651211.45
    },
    "index": 0,
    "maxPhasePower": 24000.0,
    "mode": 1,
    "power": {
      "L1": -40.0,
      "L2": 285.0,
      "L3": 354.0
    },
    "type": 1,
    "voltage": {
      "L1": 0.0,
      "L2": 0.0,
      "L3": 0.0
    }
  }
]
```

On your host, I'd expect a similar exception like the one above. If this is the case, please open a bug on pye3dc (you can put me in Cc on the issue). It would be better if you create the case there, as you'll be the one that has to validate the error, I can't reproduce it on my end.

closed 2023-09-15 by torbennehmer

neuromancer3142, 2023-09-16:
@torbennehmer: I need to apologise, as I am not a programmer and had some difficulties understanding what I need to do re. "to try this Python script on the HA server's command line, that should reproduce the issue". 

So I can describe the steps I did and perhaps you can give me feedback on how to do it better or differently:
- I installed "Pyscript Python scripting" via HACS, as I did not know how to run a Python script otherwise
- I used your code sample from above with the following input:
- USERNAME = '...' --> Login name for https://s10.e3dc.com/s10/
- PASS = '...' --> Password for login on https://s10.e3dc.com/s10/
- KEY = '...' --> RSCP password, which I set locally on my S10E Pro

After copying the python script file in the right folder I got the following debug information via the log file of Pyscript:

*** start ***
2023-09-16 10:26:37.798 ERROR (MainThread) [custom_components.pyscript.file.2023-09-12-testfile-e3dc] Exception in </config/pyscript/2023-09-12-testfile-e3dc.py> line 11:
    print(json.dumps(e3dc.get_powermeters_data(), indent=2, default=str))
                     ^
TypeError: unsupported format string passed to NoneType.__format__
2023-09-16 10:31:18.716 ERROR (MainThread) [custom_components.pyscript.file.2023-09-12-testfile-e3dc] Exception in </config/pyscript/2023-09-12-testfile-e3dc.py> line 11:
    print(json.dumps(e3dc.get_powermeters_data(), indent=2, default=str))
                     ^
TypeError: unsupported format string passed to NoneType.__format__
2023-09-16 10:31:29.959 DEBUG (MainThread) [custom_components.pyscript] reload: yaml, reloading scripts, and restarting
2023-09-16 10:31:30.254 DEBUG (MainThread) [custom_components.pyscript] No new packages to install
2023-09-16 10:31:30.294 DEBUG (MainThread) [custom_components.pyscript.eval] file.2023-09-12-testfile-e3dc: calling E3DC(1, {'username': 'xxx', 'password': 'xxx', 'ipAddress': '192.168.10.208', 'key': 'xxx'})
2023-09-16 10:31:30.439 DEBUG (MainThread) [custom_components.pyscript.eval] file.2023-09-12-testfile-e3dc: calling get_powermeters_data(, {})
2023-09-16 10:31:30.463 ERROR (MainThread) [custom_components.pyscript.file.2023-09-12-testfile-e3dc] Exception in </config/pyscript/2023-09-12-testfile-e3dc.py> line 11:
    print(json.dumps(e3dc.get_powermeters_data(), indent=2, default=str))
                     ^
TypeError: unsupported format string passed to NoneType.__format__
*** stop ***

Is there anything I misconfigured or is the behaviour as you expected? Thanks for any advice on how to proceed.

torbennehmer, 2023-09-16:
Ah, there seem to be libs missing. You know what, let's do this otherwise so that it gets easier for you: 
I'll push the next release 3.3 as soon as I can in the next 2-3 days max., I'll still have to validate compatibility with current HA  (september release) before doing so.
It will provide us with an improve diagnostics interface, which will deal with exceptions in a better way, so that you can get me a diagnostics dump again.
Also, it'll push us to the next version of the pye3dc library, which contains a fix that might just fix your problem. We can look where we go from there.

neuromancer3142, 2023-09-16:
Super, thank you very much! Looking forward to it.

PR #55 (2023-09-15): Protect diagnostic dump against arbitrary exceptions
Wrap all calls during the diagnostics gathering in an general exception handler. If one is caught, log it. This should allow us to always get at least some diagnostic data, even if the e3dc stack behaves erratically (this happens from time to time if specific units or setups return data in a way that's unexpected by pye3dc).

fixes #46
commit 1af98c6a (2023-09-15): Protect diagnostic dump against arbitrary exceptions (#55)

Wrap all calls during the diagnostics gathering in an general exception
handler. If one is caught, log it. This should allow us to always get at
least some diagnostic data, even if the e3dc stack behaves erratically
(this happens from time to time if specific units or setups return data
in a way that's unexpected by pye3dc).

fixes #46