torbennehmer/hacs-e3dc #51: Add switch for myPV control (Heating Rod)

neuromancer3142, 2023-09-02:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

E3DC S10E Pro System offers the option to directly include and control a heating rod from the company myPV. See link:

https://my-pv.com/de/produkte/ac-elwa-e/

This control is enabled in E3DC main menu --> Smart-Functions --> Smart Home --> my-PV. The S10E Pro offers the option to either define a:
- fixed time windows in which the rod will be activated and heat up the water
- start a "boost function", which enables rod heating for 15 mins and then switch off
- define a "default behaviour" for what is done with "surplus" energy, which would normally be fed into the grid (Einspeisung), i.e. instead of feeding the power in the grid, it will be used to power the heating rod instead (dynamic power demands from 0 W to 3.000 W).

The rod itself will stop heating once it reaches a defined maximum temperature, which is set via a dial-knob at the rod itself. The S10E is not involved in this (safety feature to prevent overheating, etc.).

### Describe the solution you'd like

I would like to be able to control the rod via the S10E and RSCP (if there are suitable commands existing for that - don't know that). Desired behaviour would be:

- be able to launch/start the "boost" option via RSCP (if possible a simple on/off would also be nice)
- be able to activate / de-activate certain time windows for heating rod operation, as defined on the S10E.

This will enable me to optimise my self-used PV energy. 

### Describe alternatives you've considered

Directly control the myPV heating rod as stand-alone device in Homeassistant and not using the offered S10E / myPV integration. The heating rod offers the possibility to write a certain heating demand in Watts via Modbus. However, this is only possible if the heating rod is not linked/coupled to the E3DC, as the heating rod will not accept any other Modbus commands once it has been linked to a E3DC device. 

Due to the fact that the E3DC device is doing a nice job to dynamically demand heating for its exisiting surplus energy (to be fed into the grid) in a range from 0 W to 3.000 W, I do not want to miss out on that existing feature and do it myself in a less reliable automation. Switching on/off the boost or enabling/disabling an existing time schedule on the S10E would be much easier to do.

### Additional context

[image]

neuromancer3142, 2023-09-02:
And one important thing I wanted to mention as well @torbennehmer : I really love this HACS integration. It enabled me to include my S10E Pro within Homeassistant very quickly! Very straightforward and easy to use. Happy to further support the development! Keep up the great work!

neuromancer3142, 2024-03-20:
Is there any chance that this feature request might be considered in the near future? If not I would need to look into alternative solutions. Thanks for a quick heads-up.

torbennehmer, 2024-03-22:
Hi @neuromancer3142,
I'll see what I can do, I've been quite busy recently, but I'll look if I can find a evening or two to implement this and push a beta to you. Unfortunately, I can't test this on my end - which is the true reason for not yet implementing it actually. We'll see how it works out.
Torben

neuromancer3142, 2024-03-22:
> Hi @neuromancer3142,
>
> I'll see what I can do, I've been quite busy recently, but I'll look if I can find a evening or two to implement this and push a beta to you. Unfortunately, I can't test this [...]

Happy to help with a testing whereever I can. Just let me know when you're ready. Looking very much forward to it.

Thanks a lot for your efforts. Much appreciated!

sdoe0, 2024-05-05:
Im working on this same thing rn. There are RSCP-Tags for MYPV Devices available and they seem to work with python-e3dc. 

Regarding the three surplus modes I'm not sure. 
My guess is it's the tag named 
`MYPV_DEVICE_CONTROL_MODE`.