torbennehmer/hacs-e3dc #52: 3dc_service: Error executing script. Error for if at pos 1: set_power_limits: Unkown device

svlsbrg, 2023-09-08:
### System Health details

## System Information

version | core-2023.9.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1280
Downloaded Repositories | 9

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.08.3
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 116.5 GB
disk_used | 16.0 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.3.1), Terminal & SSH (9.7.1), HassOS SSH port 22222 Configurator (0.9.1a), RaspberryMatic CCU (3.71.12.20230826), Samba share (10.0.2), Studio Code Server (5.10.1), File editor (5.6.0), Advanced SSH & Web Terminal (15.0.7), Let's Encrypt (4.12.9), NGINX Home Assistant SSL proxy (3.5.0), Samba Backup (5.2.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 1
views | 12
mode | storage

</details>

<details><summary>Miele</summary>

component_version | 0.1.15
-- | --
reach_miele_cloud | ok

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 7. September 2023 um 10:51
-- | --
current_recorder_run | 7. September 2023 um 12:59
estimated_db_size | 1547.18 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I updated the integration after a few month and removed and re-added the device afterwards. Nevertheless the service for setting power limits is not working.

### Reproduction steps

Execute the service to set limits

### Debug logs

```text
e3dc_service: If at step 1: Error executing script. Error for call_service at pos 1: set_power_limits: Unkown device ID S10-....
e3dc_service: Error executing script. Error for if at pos 1: set_power_limits: Unkown device ID S10-...

websocket_api script: Error executing script. Error for call_service at pos 1: set_power_limits: Device f50f248ad0df2249c0fef02aca790505 is no E3DC.
```

torbennehmer, 2023-09-15:
How do you call the service? I've changed the device selection at some point in the past, is this a automation script from an early version of the integration?

svlsbrg, 2023-09-15:
Hi Torben, 

it doesn't matter if I call the service manually or by automation. 

here the automation script, which I am using. It uses precalculated limits by templates:
```
alias: e3dc_service
description: ""
trigger:
  - platform: time_pattern
    minutes: /1
condition: []
action:
  - if:
      - condition: state
        entity_id: input_boolean.e3dc_auto
        state: "on"
    then:
      - service: e3dc_rscp.set_power_limits
        data:
          device_id: S10-.......
          max_charge: "{{(states('sensor.ladeleistung_opt'))|int}}"
          max_discharge: "{{(states('sensor.entladeleistung_opt'))|int}}"
    else:
      - service: e3dc_rscp.set_power_limits
        data:
          device_id: S10-.......
          max_charge: 4000
          max_discharge: 4500
mode: single

```
or direct:
[image]
```
service: e3dc_rscp.set_power_limits
data:
  device_id: f50f248ad0df2249c0fef02aca790505
  max_charge: 4000
  max_discharge: 4000
```

BR

Rene

torbennehmer, 2023-09-16:
Can you please check what the dropdown of E3DC device ID shows if you open it?

svlsbrg, 2023-09-16:
Hi Torben, 

The drop down shows the correct device:

[image]

Is it possible, that the device id cache selects a old e3dc device? I removed and re-added the device a few times in the past during updates. 

BR
Rene

torbennehmer, 2023-09-17:
Hi Rene,
that looks like an old config entry to me. In the early prototype days, the device identified itself by the serial number. I changed this, as the integration is not capable of multi-point connections (you need one integration per E3DC for authorization purposes). At one point I decided not to support multiple E3DCs in one HA installation and simplified a bit of code. This lead to a new device ID, usually "E3DC" or something like that.
My suspicion is, that your E3DC config was created by one of the original setups. Please do one thing. Drop your E3DC from HA and readd it. Just to be on the safe side (call it a hunch) I'd probably fully restart E3DC in between.
That should give you an device ID like "S10E", depending on the model of the unit, which in turn should fix the problem.
Cheers, Torben

svlsbrg, 2023-09-17:
Hi Torben, 

the reboot after deleting the device was successful. Nevertheless the device id is a very long hash, which I could only identify by switching the service within the development tools to YAML-Mode after selecting the device: f50f248ad0df2249c0fef02aca790505

The automation is working as well, if I use the long hash as device id:
```
service: e3dc_rscp.set_power_limits
data:
  device_id: f50f248ad0df2249c0fef02aca790505
  max_charge: "{{(states('sensor.ladeleistung_opt'))|int}}"
  max_discharge: "{{(states('sensor.entladeleistung_opt'))|int}}"

```

I assumed the device ID could be easier to discover e.g. "S10-722046003..."

Now it is working again. Thank you so much for your support.

BR

Rene

emmvaueff, 2023-09-17:
I also get this error earlier this day.
But after reloading the integration and restarting HA the service can be executed without error with the same (hashed) device-id as before.

torbennehmer, 2023-09-18:
@svlsbrg What does the dropdown in the UI look like, it should be something like this:
[image]
No hashed object ID should be neccessary here (actually, I'm still wondering how it works at your end using this ID, I think I'm missing some magic in HA here right now :))

@emmvaueff could you look into the dropdown shown above as well?

torbennehmer, 2023-09-18:
Forget any mail you got, I was mislead. Let me investigate further.  Instead:

Hint so far, if you configure it on the frontend, the system should show you a nice dropdown without having you to wrestle around with hashes:

[image]

torbennehmer, 2023-09-19:
This should work with 3.3, if you still experience problems, please reopen and add a current diagnostics dump (v3.3)

emmvaueff, 2023-09-19:
After installing 3.3 and restart HA i got again the error:

2023-09-19 19:35:10.983 ERROR (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Error executing script. Error for call_service at pos 1: set_power_limits: Device 21eec5d873axxxc550698df151f4eb7a is no E3DC.
2023-09-19 19:35:10.985 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [546842906176] Error handling message: set_power_limits: Device 21eec5d873axxxc550698df151f4eb7a is no E3DC. (unknown_error) Administrator from 10.0.0.227 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/117.0.0.0 Safari/537.36 Edg/117.0.2045.31)

[image]
[image]

Here is the dump:

`{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.09.2",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "miele": {
      "version": "v2021.10.12",
      "requirements": [
        "requests_oauthlib>=1.3.0"
      ]
    },
    "ecoflow_cloud": {
      "version": "0.12.1",
      "requirements": [
        "paho-mqtt==1.6.1",
        "reactivex==4.0.4",
        "protobuf>=4.23.0"
      ]
    },
    "epex_spot": {
      "version": "1.4.0",
      "requirements": [
        "bs4"
      ]
    },
    "xiaomi_miio_fan": {
      "version": "2023.7.0.0",
      "requirements": [
        "construct==2.10.56",
        "python-miio>=0.5.12"
      ]
    },
    "ecoflow": {
      "version": "2.1",
      "requirements": [
        "reactivex"
      ]
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "tapo_control": {
[... 787 lines cut ...]
              "DB_BAT_CHARGE_LEVEL",
              "Float32",
              68.65799713134766
            ],
            [
              "DB_BAT_CYCLE_COUNT",
              "Int32",
              0
            ],
            [
              "DB_CONSUMED_PRODUCTION",
              "Float32",
              76.0958480834961
            ],
            [
              "DB_AUTARKY",
              "Float32",
              69.87977600097656
            ]
          ]
        ]
      ]
    ]
  }
}`

torbennehmer, 2023-09-19:
@emmvaueff 
Hm, that is strange. I'm really at a loss here what's going on.

To be sure: Did you restart HA after removing and before readding the integration so that HA's internal configuration state starts with a clean plate? If no, please try it that way.

If yes, please check the .storage folder for the E3DC config data. I'd be interested both in the device and the integration records, especially the various object id's etc. Redact all IPs, passwords etc. from the config and send it to me via Private Mail to torben <at> nehmer <dot> net. If you want to keep the data out of the mail, use can use my personal pastebin service at bin.nehmer.net as an intermediate (hosted on my personal webserver).

emmvaueff, 2023-09-27:
Hello @torbennehmer, i'm sorry for the late response.
Your tipp "removing integration + HA-Restart + adding integration" seamed to help for a while.
Today at 3:00 inside an automation it worked last and at 14:00 without any changes in HA it now errors allways.

torbennehmer, 2023-09-28:
Could you upload the debug log around that time, preferably with enough lead time to https://owncloud.nehmer.net/index.php/s/C2tpE6faXxsXXk2 ?
Does restarting HA help?
If it fails now, is it the same error as above?
Final question: If not, please enable debug logging as well.

torbennehmer, 2023-09-28:
One other recommendation: When looking at the diagnostics dump, I can see that you're running with active power limits while setting them to the maximum allowed value. Out of my guts I'd recommend actually disabling them if you don't need them using the appropriate service, so that the "Power Limits" flag is actually false.

svlsbrg, 2023-09-28:
> Hello @torbennehmer, i'm sorry for the late response.
> 
> Your tipp "removing integration + HA-Restart + adding integration" seamed to help for a while.
> 
> Today at 3:00 inside an automation it worked last and at 14:00 without any changes in HA it now errors allways.

Hello, 

Same here. Could it be possible that the cache could be the problem?

The Device ID is correct. 

BR
Rene

svlsbrg, 2023-10-05:
> Could you upload the debug log around that time, preferably with enough lead time to https://owncloud.nehmer.net/index.php/s/C2tpE6faXxsXXk2 ?
> 
> Does restarting HA help?
> 
> If it fails now, is it the same error as above?
> 
> Final question: If not, please enable debug logging as well.

Hi Torben, 

Today I updated everything and restarted. But unfortunately without success. 

Please let me know if I can support you by debugging. 

BR
Rene

torbennehmer, 2023-10-06:
I'll try to think about some way to validate the cache and to add the important stuff to the diagnostics system to get more info. I'm still at a loss, what happens here. In the meantime, could you please check off the other things:

> Could you upload the debug log around that time, preferably with enough lead time to https://owncloud.nehmer.net/index.php/s/C2tpE6faXxsXXk2 ?
> 
> If not [enabled], please enable debug logging as well.

I'd be especially interested, as written, to the logging data around the time things broke.

torbennehmer, 2023-10-06:
If you feel saavy enough, you could try to patch `_resolve_device_id` in services.py to circumvent all local caching, resolving device on every call to rule out caching.

torbennehmer, 2023-11-07:
I have found the issue, it is related to other integrations adding additional IDs to the device (in my case a fritz.box). I'm rewriting the device detection to take this into account. Sorry for that, I did not know HA good enough to be aware that this could happen.

closed 2023-11-07 by torbennehmer

torbennehmer, 2023-11-10:
fixed with 3.4.3

PR #84 (2023-11-07): Allow for other device identifiers on services
Other integrations (like FritzBoxes) add additional identifiers to our devices. We have to take this into account when resolving device IDs to our coordinator entries, otherwise we'd incorrectly complain about our E3DC being none.

fixes #52