torbennehmer/hacs-e3dc #53: E3DC blackout times

ThomasFeistel, 2023-09-10:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

...

### Describe the solution you'd like

Hello,

I've tried out the project and I really like it! It would be great, however, if the option to set blackout times could be added. Additionally, I would appreciate it if the power meter could be integrated not only for consumption but also for additional production. I use an E3/DC S10 Pro with an additional battery pack, an E3/DC SWR123 (an external inverter from E3/DC), and one from SolarEdge. One has an internal power meter that is not currently displayed, and the other has an external power meter, also not displayed. I also have two wall chargers (not the new ones, but the ones with e3dc branding from Walbe). If you need any data or information, I am happy to provide it. Please note, though, that my programming skills are limited.

Thank you very much for your hard work on this project!
[image]

### Describe alternatives you've considered

.....

neuromancer3142, 2023-09-12:
Hi Thomansky, 
I use an S10E Pro and found the following sensor when looking through all S10 entities: "sensor.s10e_pro_additional_source_production". I simply had to unmask / enable it and then I was able to see the wattage produced by my external source, i.e. my second inverter. Perhaps this might at least partially solve your point mentioned above.
All the best ...

torbennehmer, 2023-09-12:
@Thomansky please don't file multiple feature requrests in one issue (you checked it off, did you). Also, please provide the diagnostic dump if it works on your installation (there is an issue open I don't yet understand that can break it).
As for the additional powermeters, there is an issue open already (#23, possibly along with #43 or #44 possibly).
AS for the wallbox, there too is an issue open already (#47).
Please check them first, then reopen a new issue per item missing yet.
Thanks!

closed 2023-09-12 by torbennehmer