torbennehmer/hacs-e3dc #64: Possibility to set emergency power values

hmichels, 2023-10-05:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Not really a problem, but a wish: make settings of the emergency power reserve via HA.

### Describe the solution you'd like

Changing the emergency power reserve via HA (automation).

### Describe alternatives you've considered

-

### Additional context

An introduction to the topic could be this one:
[ioBroker](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/89)

Different tags are mentioned like 
```
#define TAG_SE_REQ_SET_EP_RESERVE                             0x1B000010
#define TAG_SE_REQ_EP_RESERVE                                 0x1B000009
#define TAG_SE_EP_RESERVE                                     0x1B800009
#define TAG_SE_PARAM_INDEX                                    0x1B040000
#define TAG_SE_PARAM_EP_RESERVE                               0x1B040023
#define TAG_SE_PARAM_EP_RESERVE_W                             0x1B040033
#define TAG_SE_PARAM_EP_RESERVE_MAX_W                         0x1B040034
```
But I found also the following tags
```
EMS_REQ_EP_RESERVE = 0x01000242
EMS_EP_RESERVE = 0x01800242
```
The different numbers irritate me :-|
Now I hope that here with the HACS solution a setting of the emergency power values is possible.

Thank you very much.

ESDN83, 2026-10-23:
I looked into this and have a working implementation, tested on an S10E Pro (P10_2026_027) in HA 2026.10.2. Findings that may help, whoever implements it:

- `SE_REQ_EP_RESERVE` returns the reserve in Wh (`SE_PARAM_EP_RESERVE_W`, the value the unit displays) and the maximum (`SE_PARAM_EP_RESERVE_MAX_W`). `SE_PARAM_EP_RESERVE` and `EMS_EP_RESERVE` are the share of that maximum, not of the battery, which explains the different numbers above.
- `SE_REQ_SET_EP_RESERVE` with `SE_PARAM_INDEX` 0 and the Wh value sets it; the unit answers with the new values. All tags are already in pye3dc.

My version adds a number entity in Wh (0 to the reported maximum) plus the raw container in diagnostics. Would you like a PR for this, or would you rather build it your way?

ESDN83, 2026-10-24:
For anyone who wants to try it before an official release: the changes are on [ESDN83/hacs-e3dc, branch feature/emergency-power-reserve](https://github.com/ESDN83/hacs-e3dc/tree/feature/emergency-power-reserve). It adds the number entity *Emergency power reserve* in Wh (0 up to the maximum your unit reports).

To test it, back up `config/custom_components/e3dc_rscp`, copy the folder from the branch over it and restart Home Assistant. The branch is based on current `main`, so it also contains everything merged since v4.0.0 and needs HA 2026.10. A later HACS update replaces it again.

Be aware that a reserve above the energy currently stored makes the E3DC charge from the grid right away. If you try it, it would be great to hear your unit's values (`SE_REQ_EP_RESERVE` in the diagnostics), so we know whether other models behave the same.