torbennehmer/hacs-e3dc #65: Service "e3dc_rscp.manual_charge" only works once! Second time it is not working.

ZungeWegIchFurze, 2023-10-11:
### System Health details

## System Information

version | core-2023.10.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4795
Installed Version | 1.33.0
Stage | running
Available Repositories | 1308
Downloaded Repositories | 12

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.0
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 234.0 GB
disk_used | 21.0 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Spotify Connect (0.12.5), Terminal & SSH (9.7.1), Mosquitto broker (6.3.1), Frigate (0.12.1), Cloudflared (4.2.6), Advanced SSH & Web Terminal (15.0.7), File editor (5.6.0), InfluxDB (4.7.0), Studio Code Server (5.11.0), Piper (1.4.0), Whisper (1.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 3
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 2, 2023 at 19:46
-- | --
current_recorder_run | October 11, 2023 at 16:06
estimated_db_size | 1198.40 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hello,

the service "e3dc_rscp.manual_charge" works fine as action in automation or as service call. But only the one time. If I try to call this service again with the same command, nothing happens and the log file looks like this:

```
This error originated from a custom integration.

Logger: custom_components.e3dc_rscp.coordinator
Source: custom_components/e3dc_rscp/coordinator.py:446
Integration: E3DC Remote Storage Control Protocol (Git) (documentation, issues)
First occurred: 16:06:59 (2 occurrences)
Last logged: 16:09:18

Manual charging could not be activated
```

If I reboot the S10E, the service works again. Maybe this could also be a hardware issue.

Any Ideas? Thank you for your work, Torben!

### Reproduction steps

1. I startet charging with the service template: 
```
service: e3dc_rscp.manual_charge
data:
  device_id: 9317de1ac45a622558cc6f270f90283a
  charge_amount: 1000
```
2. I wait until 1000 watts are charged. Also I am be able to stop the charging by set a charge_amount of 0.
3. If I try, to activate the charging again with the command from bullet (1.), the service seems to be executed (green checked), but nothing happens. The debug log says: 
```
2023-10-11 15:44:02.820 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Starting manual charge of: 200
2023-10-11 15:44:02.830 WARNING (MainThread) [custom_components.e3dc_rscp.coordinator] Manual charging could not be activated
```
4. Reload or redownloading of the custom component didn't help. Nor the rebooting of Home Assistant. But the reboot of the S10E

### Debug logs

```text
2023-10-11 15:44:02.820 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Starting manual charge of: 200
2023-10-11 15:44:02.830 WARNING (MainThread) [custom_components.e3dc_rscp.coordinator] Manual charging could not be activated
```

### Diagnostics dump

-

torbennehmer, 2023-10-12:
This is a known limitation, see readme file and also the S10 Website itself. Once you initiate charging, the only thing which works is stopping the manual charge. You'll have to wait for quite some time - I don't know how long - until you can start another manual charging cycle. There is nothing I can do from my end to work around this.

closed 2023-10-12 by torbennehmer

ZungeWegIchFurze, 2023-10-12:
Thank you for your fast reply. It seems, that the E3DC has at least 2 hours of cooldown to start another manual charge due to my tests.

Keep up the good work and thank you for the great tool!