torbennehmer/hacs-e3dc #70: Manual memory charging power and memory discharging power. is broken

Thomansky, 2023-10-21:
### System Health details

Manual memory charging power and memory discharging power. is broken

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Manual memory charging power and memory discharging power. is broken

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
Traceback (most recent call last):
  File "/config/custom_components/e3dc_rscp/coordinator.py", line 471, in async_set_power_limits
    result: int = await self.hass.async_add_executor_job(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 2130, in set_power_limits
    RscpType.UInt32,
    ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: UInt32
```

### Diagnostics dump

Manual memory charging power and memory discharging power. is broken

torbennehmer, 2023-10-22:
This Bug has been introduced by the upstream library, I'll have to file an update PR there, if you need it in the meantime and have no need for the new PR code, downgrade to 3.2

torbennehmer, 2023-10-22:
https://github.com/fsantini/python-e3dc/pull/99

torbennehmer, 2023-10-22:
This will be fixed with hacs-e3dc 3.4.1, to be released shortly.

PR #72 (2023-10-22): Update pye3dc to 0.8.2
fixes #70
commit ed867bc8 (2023-10-22): Update pye3dc to 0.8.2 (#72)

fixes #70
closed 2023-10-22 by torbennehmer