torbennehmer/hacs-e3dc #73: Make diagnostics dump mo resilient

torbennehmer, 2023-10-24:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The dump should be more resilient when config keys are missing etc. It should not crash instead do its best when anonymizing and adding data. Also, catch exceptions where possible.

### Describe the solution you'd like

.

### Describe alternatives you've considered

.

### Additional context

.

### Diagnostics dump

.

torbennehmer, 2023-10-24:
@eikowagenknecht I've pulled out the troubles with the diagnostic dumping to this task, so that we can stop cluttering #45 with it. If you see any more problems in this regard, please let me know on this bug.
Can you please test your system against 3.4.2 (just released) if it works now?

eikowagenknecht, 2023-10-25:
This finally worked! Seems like all relevant places are also redacted :-)

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.10.1",
    "host_os": "Home Assistant OS 11.0",
    "docker_version": "24.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "opensprinkler": {
      "version": "1.2.3",
      "requirements": [
        "pyopensprinkler==0.7.8"
      ]
    },
    "spook_inverse": {
      "version": "1.3.1",
      "requirements": []
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "circadian_lighting": {
      "version": "2.1.4",
      "requirements": []
    },
    "spook": {
      "version": "1.3.1",
      "requirements": []
    },
    "goecharger": {
      "version": "0.25.3",
      "requirements": [
        "goecharger==0.0.16"
      ]
    },
    "gardena_smart_system": {
      "version": "1.0.0",
      "requirements": [
        "py-smart-gardena==1.3.7"
      ]
    },
    "tesla_custom": {
      "version": "3.18.0",
      "requirements": [
        "teslajsonpy==3.9.5"
      ]
    },
    "eufy_security": {
      "version": "7.7.1",
      "requirements": [
        "websocket-client==1.4.2",
        "aiortsp==1.3.7"
[... 482 lines cut ...]
              "Float32",
              24.097999572753906
            ],
            [
              "DB_BAT_CYCLE_COUNT",
              "Int32",
              0
            ],
            [
              "DB_CONSUMED_PRODUCTION",
              "Float32",
              100.0
            ],
            [
              "DB_AUTARKY",
              "Float32",
              0.0
            ]
          ]
        ]
      ]
    ]
  }
}
```

torbennehmer, 2023-11-07:
Thanks for testing, I'll close this now. Reopen this or a new one if you encounter any additional problems.

closed 2023-11-07 by torbennehmer