torbennehmer/hacs-e3dc #75: States do not correctly reply after midnight - timing issue? Winter time

Phoenix-DH, 2023-11-01:
### System Health details

I do kot know how to explain it the best. But I figured out that there might be at least an issue with the state_of_charge.

I use an automatism to send me a push when the battery is empty. And since thw last days when it is getting darker the battery is not charged 100% and gets empty in the night. I get each day a push arround the same time. Then I did a deeper look.

I hope it is visible in the screen. At 00:00 the sensor might be unavailable till 01:00.

Maybe an issue with timing? Winter time in germany?

I checked also other 24/7 states, the same.

[image]
[image]

Just let me know if I can provide additional data.

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

—

### Reproduction steps

Check other sensors and days.

### Debug logs

```text
—
```

torbennehmer, 2023-11-04:
This looks to me like a connectivity issue. Please look into the logs for that timespan, I assume that you'll find some hints there. If this happens often, you should turn on debug logging for E3DC. That'll give you more information. 
I'd rule out DST at first glance , as this time the clock was turned back, not forward. Also, the DST shift happens on 2am, not midnight. 
A diagnostic dump would be interesting as well so that I can get a better impression on your unit.

Phoenix-DH, 2023-11-04:
Which logs do you mean? The system logs?

There I get the following:
`This error originated from a custom integration.

Logger: custom_components.e3dc_rscp.coordinator
Source: helpers/update_coordinator.py:290
Integration: E3DC Remote Storage Control Protocol (Git) (documentation, issues)
First occurred: October 28, 2023 at 12:00:46 AM (2280 occurrences)
Last logged: 1:02:19 AM

Unexpected error fetching e3dc_rscp data: Max retries reached
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 518, in sendRequest
    result = self.rscp.sendRequest(request)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc_rscp_local.py", line 107, in sendRequest
    raise CommunicationError(receive[2])
e3dc._e3dc_rscp_local.CommunicationError: UNKNOWN

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/coordinator.py", line 184, in _async_update_data
    db_data_today: dict[str, Any] = await self.hass.async_add_executor_job(
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 861, in get_db_data_timestamp
    response = self.sendRequest(
               ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 529, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached
`

As mentioned, it happens dayly.

Will try to also provide a diagnostic log.

Phoenix-DH, 2023-11-05:
This night exactly the same errors in the logs, but one hour earlier after the reboot.

robelix79, 2023-11-13:
[
[config_entry-e3dc_rscp-b6b0bfa1ec5f5af18069aee162b2ae22.json](https://github.com/torbennehmer/hacs-e3dc/files/13339408/config_entry-e3dc_rscp-b6b0bfa1ec5f5af18069aee162b2ae22.json)
](url)

robelix79, 2023-11-13:
File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 237, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached
2023-11-13 18:16:47.688 ERROR (MainThread) [custom_components.e3dc_rscp.coordinator] Unexpected error fetching e3dc_rscp data: Max retries reached
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc_rscp_local.py", line 125, in connect
    self.socket.connect((self.ip, PORT))
TimeoutError: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 225, in sendRequest
    self.rscp.connect()
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc_rscp_local.py", line 130, in connect
    raise CommunicationError
e3dc._e3dc_rscp_local.CommunicationError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/coordinator.py", line 146, in _async_update_data
    poll_data: dict[str, Any] = await self.hass.async_add_executor_job(
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 303, in poll
    ts = self.sendRequestTag(RscpTag.INFO_REQ_UTC_TIME, keepAlive=True)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 263, in sendRequestTag
    return self.sendRequest(
           ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/e3dc/_e3dc.py", line 237, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached
2023-11-13 18:17:16.563 WARNING (MainThread) [homeassistant.config_entries] Config entry 'E3DC S10X' for e3dc_rscp integration could not authenticate: Max retries reached
2023-11-13 18:29:08.816 WARNING (MainThread)

torbennehmer, 2023-11-22:
@robelix79 
> TimeoutError: timed out

This looks to me as if your internal networking failure. Did it recover? If it recovers, what do the logs say?

@Phoenix-DH 
> e3dc._e3dc_rscp_local.CommunicationError: UNKNOWN

Yours is even stranger, I have no idea what a UNKNOWN comm error might be.

I have to think about it a bit, maybe adding some retry with a new connection. We're using keepalives to limit the load on the E3DC's network stack. I have to think about a graceful way to try to recover this, I don't want to just bombard the unit with retries.

Phoenix-DH, 2023-11-22:
@torbennehmer 
No worries. I habe no issue with it.
Everything is working as expected, except the automation is sending an notification …

derhappy, 2023-11-24:
I have the same disconnects. Got used to it by now, because a lot of other integrations have this kind of drop outs as well. 
I'm pretty sure it's a general issue with HA.

Phoenix-DH, 2023-11-25:
@derhappy 
Whn did you start to realize it?
I had the feeling after summer time switch.
But 2023.11 has also been released on 01.11. - can be …

What other integrations do you habe with that drop?

derhappy, 2023-11-26:
> @derhappy Whn did you start to realize it? I had the feeling after summer time switch. But 2023.11 has also been released on 01.11. - can be …
> 
> What other integrations do you habe with that drop?

I can't even remember when I started to see these issues. I think I first started noticing when I switched from a HA Supervised installation on a bare metal Ubuntu to HA Container von TrueNas Scale in January (2023). I'm pretty sure I would have noticed them before and did not just ignore them. By now I'm running HA Container on Ubuntu again, but without the unsupported supervisor and the issues persist.
Our ACs have a lot of timeouts (this seems to be an issue for many users), one of our ESPHome devises is affected, a modbus device, KNX as well as my partially home-brewed old E3DC integration using shell commands and command line sensors (HA - Host - E3DC).

That's why I think it's a general issue with HA and not specific to this integration. 
I took some time to look into the issue, but did not figure it out. Since it's usually only some sensor readings beeing lost, which I don't really need if I'm honest to myself, I stopped spending time on this matter pretty quickly.

torbennehmer, 2023-12-09:
What you may want to do is to put HA in full debug mode. At least if your underlying system's storage can handle that. It'll spew out a ton of data (really an ocean). My suspicion is that one of your integrations misbehaves. HA runs single-threaded as of now (Python can't do Multi Threading). So if one piece of code misbehaves, it can block the entire system. Usually this happens, if the integration (or whatever) blocks for some process, actively waiting instead of asynchronously doing so. E3DC does this as well (unfortunately), so while accessing the E3DC, it'll block. I could imagine that other integration do the same. If one task blocks for 30 minutes, nothing else will happen during this time.
I've this in the back of my head for E3DC, what I'd actually like to do is move pye3dc to an optional, async model. However, this is a non-trivial change to the lib, and I'm not the owner there, so this is nothing, I can change quickly.

Phoenix-DH, 2023-12-10:
Thanks for the exlanation.
Maybe I will try and habe a look.

Phoenix-DH, 2024-01-09:
Just forgot to take a look to the mentioned ...

But what I saw today by accident in the portal settings:
[image]

May that be the issue?

torbennehmer, 2024-01-09:
@Phoenix-DH That's active with my unit as well, I don't think it is related to that. It only affects the inverter, not the unit itself, the API and everything else stays reachable.

Phoenix-DH, 2024-01-09:
@torbennehmer 
Thanks for the quick reply.

torbennehmer, 2024-01-15:
@Phoenix-DH I'm closing this for now as not reproducible or possible side-effect from other integrations. Please don't hesitate to reopen the issue if you have additional information.

closed 2024-01-15 by torbennehmer

Phoenix-DH, 2024-01-15:
No problem. Will do so.
Currently I just adapted my automation.