torbennehmer/hacs-e3dc #82: sensor.s10x_maximum_charge and sensor.s10x_minimum_charge show unavailable after core update 10.5 -> 11

tschrenk, 2023-11-03:
### System Health details

## System Information

version | core-2023.11.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.59
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4755
Installed Version | 1.33.0
Stage | running
Available Repositories | 1323
Downloaded Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.1
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.1
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 457.7 GB
disk_used | 9.8 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (15.1.0), File editor (5.6.0), Log Viewer (0.16.0), Samba share (10.0.2), Zigbee2MQTT (1.33.2-1), Mosquitto broker (6.3.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 11
-- | --
resources | 4
views | 13
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 23, 2023 at 7:22 AM
-- | --
current_recorder_run | November 3, 2023 at 6:35 PM
estimated_db_size | 913.55 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

After core update in Dashboard sensor.s10x_maximum_charge and sensor.s10x_maximum_discharge.
In the Entities view these have a red status. 
This entity is no longer being provided by the e3dc_rscp integration. If the entity is no longer in use, delete it in settings.

### Reproduction steps

1. Restore old backup core <11.0
2. Maximum charge and discharge show a value
3. It is possible to change the values via automation and the call E3DC Remote Storage Control Protocol (Git): Set power limits
4. Update Core to 11
5. sensor.s10x_maximum_charge and sensor.s10x_maximum_discharge show unavailable
6. There are sensors sensor.s10x_system_maximum_charge and sensor.s10x_system_maximum_discharge but these stay at 12000 and cannot be set via Set power limits

### Debug logs

```text
Log is 18 GB on debug level therefor the tail
2023-11-03 18:38:20.111 DEBUG (MainThread) [pyipp.parser] Parsing Attribute at offset 1113
2023-11-03 18:38:20.111 DEBUG (MainThread) [pyipp.parser] Attribute Name: 
2023-11-03 18:38:20.111 DEBUG (MainThread) [pyipp.parser] Attribute Value Offset: 1118
2023-11-03 18:38:20.111 DEBUG (MainThread) [pyipp.parser] Attribute Value Length: 28
2023-11-03 18:38:20.111 DEBUG (MainThread) [pyipp.parser] Attribute Language: en
2023-11-03 18:38:20.111 DEBUG (MainThread) [pyipp.parser] Attribute Text Length: 22
2023-11-03 18:38:20.111 DEBUG (MainThread) [pyipp.parser] Attribute Value: Yellow Toner Cartridge
2023-11-03 18:38:20.111 DEBUG (MainThread) [pyipp.parser] Parsing Attribute at offset 1146
2023-11-03 18:38:20.111 DEBUG (MainThread) [pyipp.parser] Attribute Name: marker-types
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Value Offset: 1163
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Value Length: 5
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Bytes: b'toner'
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Value: toner
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Parsing Attribute at offset 1168
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Name: 
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Value Offset: 1173
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Value Length: 5
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Bytes: b'toner'
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Value: toner
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Parsing Attribute at offset 1178
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Name: 
2023-11-03 18:38:20.112 DEBUG (MainThread) [pyipp.parser] Attribute Value Offset: 1183
2023-11-03 18:38:20.113 DEBUG (MainThread) [pyipp.parser] Attribute Value Length: 5
2023-11-03 18:38:20.113 DEBUG (MainThread) [pyipp.parser] Attribute Bytes: b'toner'
2023-11-03 18:38:20.113 DEBUG (MainThread) [pyipp.parser] Attribute Value: toner
2023-11-03 18:38:20.113 DEBUG (MainThread) [pyipp.parser] Parsing Attribute at offset 1188
2023-11-03 18:38:20.113 DEBUG (MainThread) [pyipp.parser] Attribute Name: 
2023-11-03 18:38:20.113 DEBUG (MainThread) [pyipp.parser] Attribute Value Offset: 1193
2023-11-03 18:38:20.113 DEBUG (MainThread) [pyipp.parser] Attribute Value Length: 5
2023-11-03 18:38:20.113 DEBUG (MainThread) [pyipp.parser] Attribute Bytes: b'toner'
2023-11-03 18:38:20.113 DEBUG (MainThread) [pyipp.parser] Attribute Value: toner
2023-11-03 18:38:20.114 DEBUG (MainThread) [homeassistant.components.ipp.coordinator] Finished fetching ipp data in 0.797 seconds (success: True)
2023-11-03 18:38:20.173 DEBUG (MainThread) [bleak.backends.bluezdbus.manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FD_EC_E8_DC_C7_88): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -63)>}, []]
2023-11-03 18:38:20.208 DEBUG (MainThread) [bleak.backends.bluezdbus.manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_0C_5C_7D_B3_14_70): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -70)>}, []]
2023-11-03 18:38:20.216 DEBUG (MainThread) [bleak.backends.bluezdbus.manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_5C_EF_C9_E6_52_37): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -78)>, 'AdvertisingFlags': <dbus_fast.signature.Variant ('ay', bytearray(b'\x1a'))>}, []]
2023-11-03 18:38:20.230 DEBUG (MainThread) [homeassistant.components.fritz.common] Checking host info for FRITZ!Box device 192.168.1.74
2023-11-03 18:38:20.233 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [139987804318016] tschrenk from 192.168.1.40 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.0 Safari/605.1.15): Received {'type': 'auth/sign_path', 'path': '/api/error_log', 'id': 196}
2023-11-03 18:38:20.233 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [139987804318016] tschrenk from 192.168.1.40 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.0 Safari/605.1.15): Sending {"id":196,"type":"result","success":true,"result":{"path":"/api/error_log?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI5MDk3ZGE3MjhhYTc0MDIwYWY0MDE5MzQyOTdhYWM1ZiIsInBhdGgiOiIvYXBpL2Vycm9yX2xvZyIsInBhcmFtcyI6W10sImlhdCI6MTY5OTAzMzEwMCwiZXhwIjoxNjk5MDMzMTMwfQ.S-FjSNMwfp-ZpWrZSjSx_v2FzNlkiigtrOmuBrIfdEY"}}
2023-11-03 18:38:20.246 DEBUG (MainThread) [bleak.backends.bluezdbus.manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_66_73_88_F4_60_20): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -73)>}, []]
2023-11-03 18:38:20.254 DEBUG (MainThread) [homeassistant.components.fritz.common] Checking devices for FRITZ!Box device 192.168.1.74
```

tschrenk, 2023-11-04:
Tried it 3 times from a working core 10.5 configuration
Updated the device
HA restart
Downgraded to 3.4.1
HA restart
Downgraded to 3.4.0
HA restart
Deleted S10X
HA Restart -> no entity found as expected
Added S10X 
HA restart -> Fields still unavailable
pressed button to update again to 3.4.2
HA Restart 

The issue is the update from core 10.5 -> core 11
If I go back to the 10.5 backup the fields are shown and can be changed.

torbennehmer, 2023-11-04:
This looks strange. Pleas activate debug logging for the E3DC integration and send me all E3DC lines that you find (Greg for them). I also need the diagnostic dump. Ideally from both states (working and broken). With that I cannot really help you.

tschrenk, 2023-11-04:
[01 diagnostics-working-core10.5-config_entry-e3dc_rscp-b8baddf4eae538b11f6ae943a90a8cf8.json.txt](https://github.com/torbennehmer/hacs-e3dc/files/13257144/01.diagnostics-working-core10.5-config_entry-e3dc_rscp-b8baddf4eae538b11f6ae943a90a8cf8.json.txt)
[02 diagnostics-broken-core11.1-config_entry-e3dc_rscp-b8baddf4eae538b11f6ae943a90a8cf8.json.txt](https://github.com/torbennehmer/hacs-e3dc/files/13257145/02.diagnostics-broken-core11.1-config_entry-e3dc_rscp-b8baddf4eae538b11f6ae943a90a8cf8.json.txt)
[03 debug mode component home-assistant_e3dc_rscp_2023-11-04T16-14-02.822Z.log](https://github.com/torbennehmer/hacs-e3dc/files/13257146/03.debug.mode.component.home-assistant_e3dc_rscp_2023-11-04T16-14-02.822Z.log)
[04 home assistant logs download home-assistant_2023-11-04T16-15-33.643Z.log](https://github.com/torbennehmer/hacs-e3dc/files/13257147/04.home.assistant.logs.download.home-assistant_2023-11-04T16-15-33.643Z.log)
[05 e3dc-grep.txt](https://github.com/torbennehmer/hacs-e3dc/files/13257148/05.e3dc-grep.txt)

Here the 5 files:
01 Diagnostics on hacs-e3dc on core 10.5 sensors were showing up in the dashboard
Updated core 10.5 -> 11.1 (new, came uo today)
sensors were showing up in the dashboard as unavailable

02 Diagnostics on hacs-e3dc on core 11.1 (broken sensors)

03 in Devices set has-e3dc to debug mode and rebooted, this is the file

04 Went to settings-> logs and downloaded the logs

05 Result of 
cat '03 debug mode component home-assistant_e3dc_rscp_2023-11-04T16-14-02.822Z.log'|grep -i e3dc >e3dc-grep.txt

Hope I did all right, otherwise let me know.

Thanks so much for your help!

tschrenk, 2023-11-05:
I see a message in 04 about the sensor:
homeassistant.exceptions.HomeAssistantError: Entity sensor.s10x_maximum_charge cannot be added as the entity category is set to config
maximum:discharge is not mentioned.

tschrenk, 2023-11-05:
I have the config now running on my prod on NUC and DEV on Raspi4/4GB.
Was able to reproduce the issue on both systems.
After the update the 2 fields are unavailable, going to the settings of the entities and changing the Area gets the sensor showing numbers again instead of unavailable. However they are disconnected to the e3dc.

I had one window prod with core10.5 and in other window dev with 11.1
Executing  a change of maximum-charge on prod updates the fields on the prod dashboard, but in the same view on dev (core 11.1) the fields do not change.

It seems that through the core update the entities are disconnected on the dashboard.

tschrenk-corp, 2023-11-07:
Today I started from Scratch on my DR Raspi
Formatted Raspi SD
Flashed latest HA image
inserted SD card in raspy
Restored last core10.5 image -> everything looked nicely
Updated to core 11.1 -> entities again are unavailable

Had hoped the OS was the issue.

tschrenk-corp, 2023-11-07:
Restored to 10.5
Deleted Device e3dc
Deleted Service e3dc
restart home assistant
update to core 11.1
installed e3-rscp
devices -> added device S10X
checked entity -> sensor.s10x_maximum_charge not available
restart Home assist -> integration required reconfiguration -> reconfigured fritzbox
restart Home assist
Click on entity: Diese Entität wird nicht mehr von der Integration e3dc_rscp bereitgestellt. Wenn die Entität nicht mehr verwendet wird, lösche sie in den Einstellungen.
Updated max charge -> shows number 
changed max-discard updated location -> shows number
created manual rule set maximum charge 4000
executed it
values in dashboard do not change, e3dc portal also does not show a change
created manual rule set maximum discharge 1000
values in dashboard do not change, e3dc portal also does not show a change
set max charge and discharge in one command, values do not change, e3dc do not change
change value in e3dc portal, no change in HA dashboard
restart home assistant
Value not available

=> Reproduced error on another device (raspi) by flashing sdcard and restoring backup.

tschrenk, 2023-11-07:
I did an installation from scratch on the Raspi4/4GB
1. Formatted SD Card, flashed Home Assistant 64bit
2. Created user / password / location ...
3. Core 2023.11.1; Supervisor 2023.10.1; Operating System 11.1; Frontend 20231030.1
4. Installed and configured ssh
5. Installed and configures HA
6. Added repository https://github.com/torbennehmer/hacs-e3dc.git
7. Downloaded it
8. restart HA
9. Devices -> Added integration E3DDCRemote Storage Control Protocol (GIT)
10. Added S10X device
11. Devices -> Entities -> maximum charge and minimum charge: Diese Entität wird nicht mehr von der Integration e3dc_rscp bereitgestellt. Wenn die Entität nicht mehr verwendet wird, lösche sie in den Einstellungen.
12. Gebootet HA
13. Same

===> This installation was done from scratch and I was able to reproduce the issue.

So I tried the combinations:
core 10.5 + NUC + S10X: working
core 10.5 + Raspi + S10X: Working
core 11.1 + NUC + S10X: not working
core 11.1 + Raspi + S10X: not working
Install from Scratch core 11.1 + Raps + S10X: not working

Currently I have the configs
prod: NUK+core10.5+S10X: working
dev: Raspi+core11.1+S10X: not working

Something needs to be different with core 11.1 as I do not get running configs.
Can you help me to troubleshoot?

Thanks so much.

torbennehmer, 2023-11-08:
I will investigate this next, it probably has something to do with changed dependencies. I don#t use HA Core, so I'll have to see what I'm going to do here. I'll find a way. I'll let you know if I need more information. 
Regarding priorities. The two sensors should not be too important as both of them just show diagnostic values. Does the rest of the integration work? If so, I'd recommend just going ahead with the update until I've got a fix, it should not prevent you from using the integration.

tschrenk, 2023-11-08:
The important part of the integration does not work. The 'EDC Remote Storage Control Protocol (Git): Set power limits' call can be executed  but has no effect, so I am not able to keep the battery between 20% and 80% as I cannot set the limits by this call.
The rest of the fields show up in the dashboard. 

What you would need to reproduce is a Raspi4 with a minimum 16 GB SD-Card. Time to flash, install and execute HA including ssh and HCS about 45 Minutes. To make it ease, if you want to have a look to the system I can arrange a Web session. If you don't have a Raspi I could send you my current system. Just let me know if I can help.

torbennehmer, 2023-11-10:
fixed with 3.4.3

PR #86 (2023-11-10): Sensors can't be config entities, support HA 2023.11
Establishes compatibility to HA 2023.11

fixes #82
commit 2423b95c (2023-11-10): Sensors can't be config entities

Establishes compatibility to HA 2023.11

fixes #82
commit 3ee217db (2023-11-10): Sensors can't be config entities, support HA 2023.11 (#86)

Establishes compatibility to HA 2023.11

fixes #82
closed 2023-11-10 by torbennehmer