torbennehmer/hacs-e3dc #87: Total sensors: pv production / from grid / to grid / battery charge / battery discharge / (total per string, total wallbox usage)

Cheppetto, 2023-11-22:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Not a problem of the integration itself, more a problem to be able to hanlde data more easily.

I would like to create weekly / monthly / yearly sensor - thus the daily reset is not ideal

### Describe the solution you'd like

Additional Sensors - if data is provided via RSCP for above mentioned entities.

### Describe alternatives you've considered

workaround: addd attribute last period and value to sensor - then it can be done manually

### Additional context

needed for clean human readable dashboards to track, energy cycle within a household

torbennehmer, 2023-11-22:
Two quick tips, right now I don't have this on my immediate implementation plans:

1. If you use the energy dashboard of HA, you get this data for all assigned sensors.
2. You can use the Rieman sum integral integration to build your own sensors, just link them to the real-time sensors, not the daily ones. https://www.home-assistant.io/integrations/integration/

Cheppetto, 2023-11-22:
Hi,

Totally fine to have other, higher priorities, anyways would appreciate to have them in place at some point in time.

Have to look into your tips and test them to verify how exact for example the sumup will work.

TommiG1, 2026-10-23:
Partly covered.

Day totals (`db-day-*`) have been there for a while. #409 adds the same set for month and year, plus additional production energy. Wallbox and additional powermeter lifetime style totals were already available (`wallbox-*-energy-*`, `additional-*-total`).

What is still not a first class lifetime Wh sensor set for PV / grid / battery (never reset). Month/year DB stats are the closest built in option right now; Utility Meter helpers remain the usual HA approach for custom periods.

Cheppetto, 2026-11-15:
Everything in place, thanks

closed 2026-11-15 by Cheppetto