torbennehmer/hacs-e3dc #88: Config Flow not opening after download

derimpuls, 2023-11-22:
### System Health details

## System Information

version | core-2023.11.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4899
Installed Version | 1.33.0
Stage | running
Available Repositories | 1337
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.1
-- | --
update_channel | stable
supervisor_version | supervisor-2023.11.3
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 234.0 GB
disk_used | 6.0 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.8.1), Dnsmasq (1.6.0), NGINX Home Assistant SSL proxy (3.5.0), File editor (5.7.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 2
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 19. November 2023 um 10:36
-- | --
current_recorder_run | 22. November 2023 um 15:31
estimated_db_size | 140.87 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Installing this integration doesn't trigger the config flow dialog as mentioned in the installation description.

### Reproduction steps

1. Add repo to HACS
2. Install the integration
3. No config flow shows up

I'm stuck with not being able to configure the integration.
...

### Debug logs

```text
The only log entry regarding hacs-e3dc says it has been discovered and could cause issues because being a community integration. Nothing else is logged.
```

### Diagnostics dump

/

derimpuls, 2023-11-22:
I just noticed this also happens on an totally fresh haos docker container with fresh configs.

torbennehmer, 2023-11-22:
Did you explicitly add the integration using "Add integration" in the integration overview in the system config? The HACS-E3DC integration currently does not support auto-configuration.

torbennehmer, 2023-11-22:
The hacs integration dialog does only download and install the codebase.
As with any other HA integration, you need to configure it in HA as well via `/config/integrations/dashboard`

derimpuls, 2023-11-22:
Oh man. Apologies! I was not aware of this. It's my first day using homeassistant.
Thanks a ton for your support!!!
Appreciate your work!

closed 2023-11-22 by derimpuls