torbennehmer/hacs-e3dc #89: Entity sensor.s10e_additional_total from integration e3dc_rscp has state class total_increasing, but its state is negative.

derhappy, 2023-11-24:
### System Health details

## System Information

version | core-2023.11.2
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.2.0-36-generic
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1339
Downloaded Repositories | 34

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 21
views | 61
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 23. August 2023 um 16:23
-- | --
current_recorder_run | 24. November 2023 um 12:19
estimated_db_size | 12965.06 MiB
database_engine | postgresql
database_version | 15.4

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..
[... 121 lines cut ...]
2023-11-24 11:22:08.772 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling general status information
2023-11-24 11:22:08.831 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Poll power settings
2023-11-24 11:22:08.837 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling manual charge information
2023-11-24 11:22:08.842 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling additional powermeters
2023-11-24 11:22:08.861 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Skipping power metrics poll.
2023-11-24 11:22:08.861 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Finished fetching e3dc_rscp data in 0.089 seconds (success: True)
2023-11-24 11:22:18.771 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling general status information
2023-11-24 11:22:18.827 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Poll power settings
2023-11-24 11:22:18.833 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling manual charge information
2023-11-24 11:22:18.838 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling additional powermeters
2023-11-24 11:22:18.857 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Skipping power metrics poll.
2023-11-24 11:22:18.857 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Finished fetching e3dc_rscp data in 0.086 seconds (success: True)
2023-11-24 11:22:28.772 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling general status information
2023-11-24 11:22:28.831 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Poll power settings
2023-11-24 11:22:28.840 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling manual charge information
2023-11-24 11:22:28.846 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling additional powermeters
2023-11-24 11:22:28.864 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Polling today's power metrics
2023-11-24 11:22:28.864 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Midnight is 2023-11-24 00:00:00+01:00, DB query timestamp is 1700780400, applied offset: 3600
2023-11-24 11:22:28.864 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Midnight DB query timestamp is 1700784000, applied offset: 3600
2023-11-24 11:22:28.892 DEBUG (MainThread) [custom_components.e3dc_rscp.coordinator] Finished fetching e3dc_rscp data in 0.120 seconds (success: True)
```

### Diagnostics dump

[e3dc_dump.json](https://github.com/torbennehmer/hacs-e3dc/files/13462390/e3dc_dump.json)

torbennehmer, 2023-11-25:
@derhappy I've worked on improving the auto-detection of the additional powermeters. I'm not entirely sure, how this works out. Would you be able to manually install the current GIT head revision?

neuromancer3142, 2023-11-25:
I have the same error. See extract from the log:

*** start ***
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:311
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 23. November 2023 um 20:25:55 (3 occurrences)
Last logged: 01:05:12

Entity sensor.s10e_pro_additional_total from integration e3dc_rscp has state class total_increasing, but its state is negative. Triggered by state -2668.92052 with last_updated set to 2023-11-23T19:19:59.999999+00:00. Please create a bug report at https://github.com/torbennehmer/hacs-e3dc/issues
Entity sensor.s10e_pro_root_total from integration e3dc_rscp has state class total_increasing, but its state is negative. Triggered by state -5334.95372 with last_updated set to 2023-11-23T19:19:59.999999+00:00. Please create a bug report at https://github.com/torbennehmer/hacs-e3dc/issues
Entity sensor.s10e_pro_energy_charged_from_grid from integration e3dc_rscp has state class total_increasing, but its state is negative. Triggered by state -0.0001785990625 with last_updated set to 2023-11-25T00:00:08.847498+00:00. Please create a bug report at https://github.com/torbennehmer/hacs-e3dc/issues
*** stop ***

torbennehmer, 2023-11-26:
Please try the current RC1:
[image]
I've just pubished it as a prerelease, so HACS can download it.

If it still fails, please always include Debug Log and the current Diagnostics Dump. If using additional powermeters, please try to take the dump when they show anything. E.g. for an additional inverter, take the dump when it is actually delivering (or draining) anything etc.

closed 2023-11-26 by derhappy