torbennehmer/hacs-e3dc #90: Total wallbox usage missing

derhappy, 2023-11-24:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

There is an entity for wallbox consumption (kW), but the entity for the total consumption (kWh) is missing. 

This is the only entity missing for me to switch from my previous e3dc interface to this integration. There is also more detailed data available (kWh solar / net), but for me the sum is already enough.

### Describe the solution you'd like

A new entity for wallbox_consumption_total (or _today) is available.

### Describe alternatives you've considered

A Riemann integral would technically be possible. 
Using another interface to get the data like before would be the another valid option.

### Additional context

Currently I'm using the go-rscp command line utility by @spali (https://github.com/spali/go-rscp/). I can get the wallbox data with these keys:

```[
  [ # day current
    "DB_REQ_HISTORY_DATA_DAY",
    [
      ["DB_REQ_HISTORY_TIME_START", some nonsense ],
      ["DB_REQ_HISTORY_TIME_INTERVAL", more nonsense ],
      ["DB_REQ_HISTORY_TIME_SPAN", guess ]
    ]
  ],
  [
    "WB_REQ_DATA",
    [
      ["WB_INDEX", 0],
      "WB_REQ_ENERGY_ALL",
      "WB_REQ_ENERGY_SOLAR"
    ]
  ]
]
```

example result
```
result:
  DB_SUM_CONTAINER:
    DB_GRAPH_INDEX: -1
    DB_BAT_POWER_IN: 799.5
    DB_BAT_POWER_OUT: 731.5
    DB_DC_POWER: 4714
    DB_GRID_POWER_IN: 57
    DB_GRID_POWER_OUT: 57612
    DB_CONSUMPTION: 63710
    DB_PM_0_POWER: 0
    DB_PM_1_POWER: -1874.5
    DB_BAT_CHARGE_LEVEL: 17.696
    DB_BAT_CYCLE_COUNT: 0
    DB_CONSUMED_PRODUCTION: 99.07392
    DB_AUTARKY: 9.571495
  WB_INDEX: 0
  WB_ENERGY_ALL: 60670
  WB_ENERGY_SOLAR: 599
friendly_name: e3dc_today
```

### Diagnostics dump

[e3dc_dump.json](https://github.com/torbennehmer/hacs-e3dc/files/13462430/e3dc_dump.json)

derhappy, 2023-11-24:
I think this data is still missing from the library you're using, so I updated a feature request there with some additional data.

torbennehmer, 2023-12-09:
@derhappy I'll look into this. One thing, could you send me a diagnostics dump when your WB is loading? I've not yet seen how the existing API reacts if the WB is active. Please attach it to #54. That would be cool!

T1ppes, 2024-04-28:
Wallbox Interactions have been merged to python-e3dc today.
https://github.com/fsantini/python-e3dc/pull/116

MBerchtold21, 2024-06-16:
I would need exactly the same. The total wallbox consumption is a important figure.
Would be really great if it is possible to add it. 

Can we expect it to be added? Or what is the plan? Thank you very much!

bullitt186, 2024-08-04:
@derhappy, @MBerchtold21 ,
[The Beta of 3.8](https://github.com/torbennehmer/hacs-e3dc/releases/tag/3.8.0-beta.5) brings improved Wallbox support, including wallbox consumption broken down into total, grid and sun:

[image]

Feel free to test and provide feedback in [this issue](https://github.com/torbennehmer/hacs-e3dc/issues/47)

MBerchtold21, 2024-08-12:
> @derhappy, @MBerchtold21 , [The Beta of 3.8](https://github.com/torbennehmer/hacs-e3dc/releases/tag/3.8.0-beta.5) brings improved Wallbox support, including wallbox consumption broken down into total, grid and sun:
> 
> <img alt="image" width="306" src="https://private-user-images.githubusercontent.com/24450990/354923197-42f1d3f4-19a8-4be3-9417-d89420004e37.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0NzA3MzksIm5iZiI6MTcyMzQ3MDQzOSwicGF0aCI6Ii8yNDQ1MDk5MC8zNTQ5MjMxOTctNDJmMWQzZjQtMTlhOC00YmUzLTk0MTctZDg5NDIwMDA0ZTM3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEyVDEzNDcxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM2YWM4ZGVkYzgzMWRiZDY2OWQ5NjZjNzEzM2ZjMGY0ODcyYzg2MzI0ZDQ2MTI3ZThlZjZhYjc2YjJlNjdlZTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IpcaElOicwOCSl56wzUDt0e8CJIPwWfP0TvhkppUo2U">
> Feel free to test and provide feedback in [this issue](https://github.com/torbennehmer/hacs-e3dc/issues/47)

Very cool, thanks a lot! Stupid question: How can I switch to the new beta version in HA?

TommiG1, 2026-10-23:
This should already be available.

Wallbox energy totals are exposed as `wallbox-*-energy-all`, plus `energy-net` and `energy-sun` (Wh, shown as kWh). Instantaneous `wallbox-consumption` is the power sensor. If those entities are missing, check disabled entities on the wallbox device, or share a diagnostics dump.