torbennehmer/hacs-e3dc #92: Additional production shows negative Values

ChristianRingwald, 2023-11-29:
### System Health details

## System Information

version | core-2023.11.3
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.0-rpi6-rpi-v8
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1348
Downloaded Repositories | 3
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 12 (bookworm)
-- | --
update_channel | stable
supervisor_version | supervisor-2023.11.5
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 28.7 GB
disk_used | 13.5 GB
healthy | true
supported | failed to load: Unsupported
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.14.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 23. November 2023 um 10:02
-- | --
current_recorder_run | 29. November 2023 um 10:13
estimated_db_size | 24.28 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

When using this integration with my E3DC S10PRO and an external solar inverter, the additional production is showing negative values. 

[image]
(Here the issue is already fixed, and entities show the correct values)

The powermeter used for additional production is configured that way in my system.
[image]

When using `get_powermeters()` directly with `python-e3dc` I see following powermeter config:
```
[
    {
        "index": 0,
        "type": 1,
        "typeName": "PM_TYPE_ROOT"
    },
    {
        "index": 1,
        "type": 2,
        "typeName": "PM_TYPE_ADDITIONAL"
    }
]
```

**Now there are two issues:**

1.  `PM_TYPE_ADDITIONAL` is not included in the match statement and therefore the values are not negated
2. The [match statement](https://github.com/torbennehmer/hacs-e3dc/blob/9a46590d4328aa9beac37c3d9d098cfee9934d49/custom_components/e3dc_rscp/coordinator.py#L123) does not work unless the enum class is used for comparing.  

### Reproduction steps

1. Have an E3DC system with additinal solar inverter and powermeter configured as type 0x02 (PM_TYPE_ADDITIONAL)
2. Install this integration 
3. Entities `additional` and `additional_total` show negative values

### Debug logs

```text
not available
```

ChristianRingwald, 2023-11-29:
@torbennehmer: I tried to push the fix, but access denied.

torbennehmer, 2023-11-29:
@ChristianRingwald This is probably a duplicate of #89. Please try [3.5.0-rc1](https://github.com/torbennehmer/hacs-e3dc/releases/tag/v3.5.0-rc.1), the current beta release. It should fix the problem. If not, please add debug logs and diagnostic dumps, without them I can't help you (all crystal balls are currently under maintenance).

> @torbennehmer: I tried to push the fix, but access denied.

Contributions are always welcome: Create a PR from a fork, you can't push to my repo directly.