torbennehmer/hacs-e3dc #96: Emergency status - request

chibbez, 2023-12-28:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

no, I am missing an sensor value for emergency status / Notstrommodus.

### Describe the solution you'd like

I would like to SEE the emergency status within HASS. This sensor would help me to run some automations in case of activated emergency state.

Based on the E3DC documentation, this value can also be SET to force the e3dc to run a test. This part is optional / only nice to have for me.

Here is the copy / paste from the docu:
1 = Notstrommodus oder Inselbetrieb ist aktiv
0 = Normalbetrieb
Üblicherweise schaltet das S10 selbstständig in den Notstrommodus wenn dies erforderlich ist.
Durch Schreiben dieses Datenpunktes kann das S10 aktiv in den Inselbetrieb bzw. zurück in den Normalbetrieb gebracht werden. Dadurch wird das S10 vom Stromnetz getrennt bzw. wieder verbunden.

### Describe alternatives you've considered

-

Michel1973, 2024-01-11:
This would be great to use for automation

torbennehmer, 2024-01-15:
This is a duplicate of #33 

Be aware, that due to probable authentication problems due to a lost internet connection, you won't be able to react immediately to a power outage, probably never if you don't have an appropriate network fallback. Local connections during internet outages are unsupported and I can't tell how long the keepalive will help here.

Also, be aware that on a power loss, the transition to emergency power mode takes 1-2 seconds (a relay has to shift), so your HA will loose power and first has to reboot anyway (breaking any kept-alive connection) unless your HA is buffered by an UPS. 

While get_power_status will yield all the flags, testing its reaction during a real power outage is hard, due to the repeated power losses. I'll probably have to do this when wife and kids are not at home :-D