torbennehmer/hacs-e3dc #102: E3DC Manual charge

Thomansky, 2024-01-21:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Please integrate this feature so that it would be possible to load the memory manually via Tibber, for example

[image]

### Describe the solution you'd like

.

### Describe alternatives you've considered

.

### Additional context

Wanted to ask if you can support the project financially or when the next update will come with the wallboxes

torbennehmer, 2024-01-21:
That's already implemented. See [Initate manual battery charging
](https://github.com/torbennehmer/hacs-e3dc?tab=readme-ov-file#initate-manual-battery-charging) in the Readme, read carefully, it has a few cavets.

> Wanted to ask if you can support the project financially or 

The hindering factor for me is time, not money :-)

> when the next update will come with the wallboxes

The next update will have HASS 24.01 support and a number of minor fixes, 24.01 brings a number of features, that allow me to improve error handling a bit. Next thing will be the system status support (as this is a quick-win). Wallbox will probably come next, as it is a more time-consuming feature as I don't have an E3DC wallbox (I'm using an OpenWB instead).

closed 2024-01-21 by torbennehmer

Thomansky, 2024-01-21:
Is there anything else that can help? I can't program, but I have two E3DC wallboxes. do you speak german?