torbennehmer/hacs-e3dc #103: Error: Setup failed for 'e3dc_rscp' since upgrading to 3.5

Michel1973, 2024-01-21:
### System Health details

## System Information

version | core-2024.1.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.71-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1377
Downloaded Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Discovergy</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.4
-- | --
update_channel | stable
supervisor_version | supervisor-2023.12.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 30.8 GB
disk_used | 4.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.8.1), Z-Wave JS (0.4.3), File editor (5.7.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 11. Januar 2024 um 19:34
-- | --
current_recorder_run | 21. Januar 2024 um 20:52
estimated_db_size | 245.37 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I started the update process from the HACS integration. I get the following error after the system reboot.

Error: Setup failed for 'e3dc_rscp' : Integration not found

### Reproduction steps

1. Go to HACS 
2. Start update process
3. Reboot system

### Debug logs

```text
Sorry couldn't get the debugging logs

2024-01-21 20:52:46.408 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration e3dc_rscp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-21 20:52:46.408 ERROR (SyncWorker_4) [homeassistant.loader] The custom integration 'e3dc_rscp' does not have a valid version key (untagged-a66f76cbe32ae9999176) in the manifest file and was blocked from loading. See https://developers.home-assistant.io/blog/2021/01/29/custom-integration-changes#versions for more details
2024-01-21 20:52:46.410 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration awattar_energy_cost which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-21 20:52:46.411 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-21 20:52:54.037 ERROR (MainThread) [homeassistant.setup] Setup failed for 'e3dc_rscp': Integration not found.
```

goebelmeier, 2024-01-21:
+1, can reproduce

Thomansky, 2024-01-21:
+1, can reproduce

Matthias1111, 2024-01-21:
+1, can reproduce

torbennehmer, 2024-01-22:
Pleas try if reloading the entire repo helps. I made an error when creating the release. It was fixed quickly, but apparently not quick enough. 
I beg your pardon here.

Thomansky, 2024-01-22:
I completely reloaded the project but unfortunately this didn't help

Michel1973, 2024-01-22:
I have reloaded version 3.5. But this did not help. So I reinstalled the previous version.  This is running now. 

I will retry the update to 3.5.

torbennehmer, 2024-01-22:
I've just repushed 3.5.0. to 3.5.1, i hope that this fixes things, could you check if you can cleanly move to this release?

bmause, 2024-01-22:
> Ich habe gerade 3.5.0. auf 3.5.1 zurückgeworfen, ich hoffe, dass dies die Dinge behebt, könnten Sie überprüfen, ob Sie sauber zu dieser Version wechseln können? 

gerade mit 3.5.1 upgedatet, geht wieder :-)

Thomansky, 2024-01-22:
Ja, geht wieder alles

Matthias1111, 2024-01-22:
Confirmed as well. Upgrade from 3.4.3 to 3.5.1 is working now (I rolled back from 3.5.0 to 3.4.3 yesterday.

Michel1973, 2024-01-22:
Confirm that this fix is working. Thank you for your quick support

torbennehmer, 2024-01-22:
Thanks for your patience!

closed 2024-01-22 by torbennehmer