torbennehmer/hacs-e3dc #104: Add temperatur sensor from E3DC System

Taddymalone, 2024-01-24:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I want to see when I need to heat the room in which the E3DC is standing. The external temperature is deviating from the temperature inside the E3DC System. I would like to use the internal temeratur sensor as a trigger for a heating system 

### Describe the solution you'd like

I would like to have a temperatur sensor, which can be seen on the webportal in the section  "System Monitor".

### Describe alternatives you've considered

There are no alternatives, only not to implement this

bullitt186, 2024-08-04:
Python_e3dc's function `get_pvi_data`contains already the temperature(s), so technically it's easy to integrate.
However i get on my S10X four temperatures reported back, see below.

```JSON
{
   "temperature":{
      "max":130.0,
      "min":-30.0,
      "values":[
         36.29999923706055,
         36.900001525878906,
         38.70000076293945,
         38.29999923706055
      ]
   }
}
```

My S10X Compact does not show device Temperatures in the old and/or new portal.
I find battery temperatures showing min / max but  even the max value is too low compared to the ones reported via RSCP. (+ the range is also off)

[image]
 

Any ideas what these 4 temperatures are? Four sensors? Last four values? Battery Temperatures with an offset?

Taddymalone, 2024-08-04:
Hi,

 

I can find the temperatur min. and max. in the old portal. The view "System monitor".

You can find the screen shoot attached.

 

I think the listed temperatures max and min are no actual themp. Maybe the temp range of the sensor?

 

I have a S10 pro. May be there are more sensors included....

 
 

Gesendet: Sonntag, 04. August 2024 um 21:49 Uhr

 

image.png (view on web)

TommiG1, 2026-10-23:
Implemented in #408.

PVI temperatures are exposed as `pvi-temperature-1..N` (count discovered at connect). On my S10E Pro that is four sensors around 32 to 35 °C. The same PR also adds inverter AC voltage/current/frequency and per phase powermeter energy.

closed 2026-10-25 by torbennehmer