torbennehmer/hacs-e3dc #112: Additional sensors for string 1 & 2 as well as L1,2,3 of relevant external source

wittimagic, 2024-02-26:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

No

### Describe the solution you'd like

It would be nice to have dedicated sensors for all solar fields for monitoring purposes. 
The modbus gives separate values for the connected strings (string 1 and string 2) as well as for the external generator per phase (L1,2,3).
I have individual single phase generators connected to the external generator meter and can therefore monitor each individual pv field as well as both strings directly connected.

This might be a nice feature for others too and should be considered as nice to have.

### Describe alternatives you've considered

Alternative is to not add the sensors with the addon and get them traditionally via modbus individually. 

### Additional context

This is my modbus yaml as an example:

- name: "E3DC"
  type: tcp
  host: 192.168.1.5
  port: 502
  retry_on_empty: true
  sensors:
    - name: E3DC String 1 Power
      unique_id: e3dc_string_1_power
      data_type: uint16
      address: 40101
      device_class: power
      unit_of_measurement: W
      scan_interval: 2
    - name: E3DC String 2 Power
      unique_id: e3dc_string_2_power
      data_type: uint16
      address: 40102
      device_class: power
      unit_of_measurement: W
      scan_interval: 2
    - name: E3DC Powermeter 3 L1
      unique_id: e3dc_powermeter_3_L1
      data_type: int16
      address: 40117
      device_class: power
      unit_of_measurement: W
      scan_interval: 2
    - name: E3DC Powermeter 3 L2
      unique_id: e3dc_powermeter_3_L2
      data_type: int16
      address: 40118
      device_class: power
      unit_of_measurement: W
      scan_interval: 2
    - name: E3DC Powermeter 3 L3
      unique_id: e3dc_powermeter_3_L3
      data_type: int16
      address: 40119
      device_class: power
      unit_of_measurement: W
      scan_interval: 2

wittimagic, 2024-02-26:
If you consider implementing this idea, it might be nice to add the data as additional attributes to the already given sensors.
That way the list of all sensors will not become too crowded.

torbennehmer, 2024-03-08:
Please add the diagnostic dump

wittimagic, 2024-03-08:
The dump of your the e3dc integration:
[config_entry-e3dc_rscp-7a29928e2c66dc30bb06b0ae4364c0cf.json](https://github.com/torbennehmer/hacs-e3dc/files/14540900/config_entry-e3dc_rscp-7a29928e2c66dc30bb06b0ae4364c0cf.json)

Not sure how to get a dump of the modbus internal integration.

torbennehmer, 2024-03-09:
@wittimagic I do not need a modbus dump, it#s irrelevant, as I#m not using modbus. The one of me is perfectly fine. Thanks!

TommiG1, 2026-10-23:
This should be covered by the open PRs.

Per string DC metrics (string 1 / 2 power, voltage, current, energy) are in #407. Per phase values for the external / additional source (and related AC phase energy) are in #408. Both were live tested on an S10E Pro.

Cheppetto, 2026-11-15:
Looks good