torbennehmer/hacs-e3dc #113: Additional Sensor Total wrong values E3DC X20 PRO

leodale, 2024-02-28:
### System Health details

## System Information

version | core-2024.2.4
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 5.10.0-21-amd64
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1402
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2024.02.0
agent_version | 1.4.1
docker_version | 23.0.0
disk_total | 232.2 GB
disk_used | 14.9 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.9.0), Studio Code Server (5.15.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 1
views | 39
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 19. Februar 2024 um 11:50
-- | --
current_recorder_run | 27. Februar 2024 um 15:45
estimated_db_size | 1295.17 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The Sensor sensor.s10x_additional_total provides values betweeen 0.007500 to 0.007535 during the day. The current addtional power sensor is working. Same as the other E3DC values. 

### Reproduction steps

Just set up an E3DC S20 PRO with Additional source

### Debug logs

```text
to long
```

### Diagnostics dump

empty

torbennehmer, 2024-03-08:
Can you please add a diagnostic dump

leodale, 2024-03-21:
> diagnostic dump

see main post
thank you

torbennehmer, 2024-03-22:
I need the diagnostic dump of the e3dc compontent to get further information, it looks as if the diag dump is from the HACS component.

apart from that I've forwarded this upstream to fsantini/python-e3dc#115, I'm not yet sure what the root cause could be.

leodale, 2024-03-22:
Sorry i cant download the diagnostic dump for e3dc, the Browser (Chrome) Reports "Website not available"

leodale, 2024-03-25:
Hello,
Is that the right file ?

[config_entry-e3dc_rscp-b73e3914bfddaa2b273b4b681fbf5389.json](https://github.com/torbennehmer/hacs-e3dc/files/14740522/config_entry-e3dc_rscp-b73e3914bfddaa2b273b4b681fbf5389.json)

leodale, 2024-03-25:
The bug:
Additional total (kWh) has a wrong value (that is always 0,01)

[image]

torbennehmer, 2024-03-29:
@leodale I finally got some time to investigate this nightly outage, I can see it on my end as well, seems to be some problem with the data flip to the next day. This should be solvable, but could take some time as I might have to wait for midnight with each test. I'm on it.

torbennehmer, 2024-03-29:
@leodale I will update a new beta release shortly, which I'm using on my end to verify that we react more gracefully to nightly outages. It seems as if E3DC is unable to answer DB Data dumps at midnight. I've to play around with this a bit to get a better idea, but this should no longer throw the entire integration off track.

Regarding the 0,01 kWh value: I assume that the Diagnostic dump has been taken at the same time as the screenshot? Is that the case?

torbennehmer, 2024-03-29:
@leodale If you still have this midnight only outages, please follow them on #130, not here. Let us keep this issue at the wrong total sensor value you see for your unit.

torbennehmer, 2024-05-28:
> @leodale I will update a new beta release shortly, which I'm using on my end to verify that we react more gracefully to nightly outages. It seems as if E3DC is unable to answer DB Data dumps at midnight. I've to play around with this a bit to get a better idea, but this should no longer throw the entire integration off track.
> 
> Regarding the 0,01 kWh value: I assume that the Diagnostic dump has been taken at the same time as the screenshot? Is that the case? If not, please provide me with a new dump and a screenshot from the same time (plus minus a few minutes)

Could you please update the issue accordingly?

leodale, 2024-12-12:
[image]

[home-assistant_e3dc_rscp_2024-12-12T20-36-09.030Z.log](https://github.com/user-attachments/files/18117262/home-assistant_e3dc_rscp_2024-12-12T20-36-09.030Z.log)

@torbennehmer The issue is still there. The power sensor is working but the energy sensor is not working.

[image]

leodale, 2024-12-12:
@torbennehmer 
[config_entry-e3dc_rscp-b73e3914bfddaa2b273b4b681fbf5389 (3).json](https://github.com/user-attachments/files/18117331/config_entry-e3dc_rscp-b73e3914bfddaa2b273b4b681fbf5389.3.json)

gamer123, 2024-12-12:
It looks like the value is correct adobted by the integration.
"additional-1-total": 35.78,  it is the shown value at the picture 0,04 kwh  0,0357 is round up to 0,04. So the problem must be in a lower level :|

gamer123, 2024-12-12:
it should be more than 32460.77+3421,87 kwh =35882,64 kwh so it could be also a  shifted bei 10.000.000

gamer123, 2024-12-12:
The installed Battery capacity is also different installed in this case 42kwh system tell´s 62 kwh and in homeassistant entiy it is shown as 0,06200000001. This is the same problem with my S10X Compact here is 0,033 in homeassistant shown but i can only use 11kwh. The installed battery size is 14 kwh.