torbennehmer/hacs-e3dc #114: 16777278 is not a valid RscpTag

kobelka, 2024-03-01:
### System Health details

## System Information

version | core-2024.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.6.16-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4996
Installed Version | 1.34.0
Stage | running
Available Repositories | 1401
Downloaded Repositories | 41

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.02.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 491.4 GB
disk_used | 118.7 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Node-RED (17.0.7), FTP (5.0.1), InfluxDB (5.0.0), MariaDB (2.6.1), Studio Code Server (5.15.0), Z-Wave JS UI (3.4.0), Mosquitto broker (6.4.0), ha-sip (3.5.1), Home Assistant Google Drive Backup (0.112.1), Zigbee2MQTT (1.35.3-1), Music Assistant BETA (2.0.0b106), Cloudflared (5.1.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 23
views | 59
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 31. Januar 2024 um 15:35
-- | --
current_recorder_run | 29. Februar 2024 um 18:01
estimated_db_size | 40742.48 MiB
database_engine | mysql
database_version | 10.6.12

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The addon work for a long time, but now I can´t get it connected again. I restarted the e3dc, home assistant and reinstalled the addon allready. I double checked the password etc.

### Reproduction steps

no connection to the e3dc

### Debug logs

```text
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.e3dc_rscp.e3dc_proxy
Source: custom_components/e3dc_rscp/e3dc_proxy.py:29
Integration: E3DC Remote Storage Control Protocol (Git) (documentation, issues)
First occurred: 29. Februar 2024 um 18:01:51 (608 occurrences)
Last logged: 07:01:13

Communication error with E3DC: Max retries reached
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc_rscp_local.py", line 102, in sendRequest
    receive = self._receive()
              ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc_rscp_local.py", line 76, in _receive
    decData = rscpDecode(self.encdec.decrypt(data))[0]
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_rscpLib.py", line 263, in rscpDecode
    return rscpDecode(rscpFrameDecode(data)[0])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_rscpLib.py", line 279, in rscpDecode
    innerData, usedLength = rscpDecode(data[curByte:])
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_rscpLib.py", line 270, in rscpDecode
    strTag = getStrRscpTag(hexTag)
             ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_rscpTags.py", line 3692, in getStrRscpTag
    tag = RscpTag(tag)
          ^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/enum.py", line 744, in __call__
    return cls.__new__(cls, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/enum.py", line 1158, in __new__
    raise ve_exc
ValueError: 16777278 is not a valid RscpTag

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc.py", line 226, in sendRequest
    result = self.rscp.sendRequest(request)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc_rscp_local.py", line 108, in sendRequest
    raise CommunicationError
e3dc._e3dc_rscp_local.CommunicationError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/e3dc_rscp/e3dc_proxy.py", line 29, in wrapper_handle_e3dc_ex
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/e3dc_proxy.py", line 121, in get_manual_charge
    data = self.e3dc.sendRequest(
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc.py", line 237, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.e3dc_rscp.coordinator
Source: helpers/update_coordinator.py:313
Integration: E3DC Remote Storage Control Protocol (Git) (documentation, issues)
First occurred: 29. Februar 2024 um 18:01:51 (608 occurrences)
Last logged: 07:01:13

Unexpected error fetching e3dc_rscp data: Communication Failure: Failed to send data
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc_rscp_local.py", line 102, in sendRequest
    receive = self._receive()
              ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc_rscp_local.py", line 76, in _receive
    decData = rscpDecode(self.encdec.decrypt(data))[0]
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_rscpLib.py", line 263, in rscpDecode
    return rscpDecode(rscpFrameDecode(data)[0])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_rscpLib.py", line 279, in rscpDecode
    innerData, usedLength = rscpDecode(data[curByte:])
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_rscpLib.py", line 270, in rscpDecode
    strTag = getStrRscpTag(hexTag)
             ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_rscpTags.py", line 3692, in getStrRscpTag
    tag = RscpTag(tag)
          ^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/enum.py", line 744, in __call__
    return cls.__new__(cls, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/enum.py", line 1158, in __new__
    raise ve_exc
ValueError: 16777278 is not a valid RscpTag

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc.py", line 226, in sendRequest
    result = self.rscp.sendRequest(request)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc_rscp_local.py", line 108, in sendRequest
    raise CommunicationError
e3dc._e3dc_rscp_local.CommunicationError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/e3dc_rscp/e3dc_proxy.py", line 29, in wrapper_handle_e3dc_ex
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/e3dc_proxy.py", line 121, in get_manual_charge
    data = self.e3dc.sendRequest(
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc.py", line 237, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 313, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/coordinator.py", line 157, in _async_update_data
    data = await self.hass.async_add_executor_job(self.proxy.get_manual_charge)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/e3dc_proxy.py", line 37, in wrapper_handle_e3dc_ex
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Communication Failure: Failed to send data
```

kobelka, 2024-03-01:
Version vom E3DC ist P10_2023_06

kobelka, 2024-03-04:
I did a [pull request](https://github.com/fsantini/python-e3dc/pull/108)  in the repository of [python-e3dc](https://github.com/fsantini/python-e3dc). 
Is this the right way ?

gamer123, 2024-03-05:
Same Problem here since the update to 2023_06 it stoped working.

torbennehmer, 2024-03-05:
> I did a [pull request](https://github.com/fsantini/python-e3dc/pull/108) in the repository of [python-e3dc](https://github.com/fsantini/python-e3dc). Is this the right way ?

No, not in this way, I meant filing an issue there, as this error is coming from the pye3dc layer. However, I don't really understand it so far, so please stick for it for the moment.

torbennehmer, 2024-03-05:
See https://github.com/fsantini/python-e3dc/issues/109

torbennehmer, 2024-03-05:
@kobelka I'll try to push out an hotfix release tomorrow or on thursday if possible that disables the call temporarily. That'll disable the manual charging sensors until we get a better result here. Look at the issue at fsantis repo linked above. Maybe we need your help there

torbennehmer, 2024-03-07:
@kobelka, @gamer123, @2nd-skys

I'm just pushing out 3.6.1-beta.1, it should catch the exception on the manual charging call. Right now, this is only a workaround, it will just replace the API call with dummy values, essentially silently ignoring the error. Initiating manual charging should work (unless they messed around with that too), but you won't see it anymore in the sensors (active / energy).

Please give it a try, my E3DC is still on the earlier version, so I cannot validate this on my end.

gamer123, 2024-03-07:
@torbennehmer hi thanks in my case it is working again. i havn`t tried manual charge bevor the update and also not with the workaround. I do have Tibber (dynamic prices) but in my point of view is it not usefull because the better way is to store energy in the battery for high price periods and get cheap energy during the nights for standby consumption.

2nd-skys, 2024-03-07:
@torbennehmer thank you for the quick fix! for me the integration is also working again. But I also never tried manual charge before, therefore I can't really tell any difference.

kobelka, 2024-03-08:
@torbennehmer the connection works but I am missing these Sensors

sensor.s10e_pro_transfer_to_from_grid
sensor.s10e_pro_battery_net_change

torbennehmer, 2024-03-08:
Can you please attach debug logs and an diagnostic dump ? Ideally while your battery is charging and you have a net load on the grid. If you dont‘t have both, send two diagnostic dumps.

kobelka, 2024-03-08:
It is solved, I did a fresh install of the integration and after that i saw these sensors (but they were deactivated) 
After the activation they work again. Thanks for your help.

torbennehmer, 2024-04-17:
I've pushed a PR to  https://github.com/fsantini/python-e3dc/pull/117, I think I've identified the tag (it contains the target Wh amount that has been sent with the last load request)