torbennehmer/hacs-e3dc #120: Add Sensors for extra power meters

kobelka, 2024-03-08:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

It would be nice if the external power meters are also aviable as the other sensors

### Describe the solution you'd like

Add the external power meter, like in https://github.com/rxhan/RSCPGui

### Describe alternatives you've considered

nothing

torbennehmer, 2024-03-08:
Please add a diagnostic dump, can't do without it.

kobelka, 2024-03-09:
I can´t download it:

[image]

torbennehmer, 2024-03-09:
What does the debug log say (enable debug logging for the integration before you try it, there's probably an exception flying around. Normally, the diagnostic dumper should catch this and take care of it, I'm wondering, what's actually going on here.

wittimagic, 2024-03-09:
Ok, another try:
[config_entry-e3dc_rscp-7a29928e2c66dc30bb06b0ae4364c0cf (2).json](https://github.com/torbennehmer/hacs-e3dc/files/14546568/config_entry-e3dc_rscp-7a29928e2c66dc30bb06b0ae4364c0cf.2.json)

wittimagic, 2024-03-09:
Hmm, had a different integration in the log throwing errors and some devices that can't be loaded because they are offline.
I cleaned up some stuff.

Another try:
[config_entry-e3dc_rscp-7a29928e2c66dc30bb06b0ae4364c0cf (4).json](https://github.com/torbennehmer/hacs-e3dc/files/14546669/config_entry-e3dc_rscp-7a29928e2c66dc30bb06b0ae4364c0cf.4.json)

kobelka, 2024-03-09:
@torbennehmer makes no difference...only when I exit debug mode again does the debug log appear.
Here is the last section:

`2024-03-09 16:34:27.471 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 91, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 25, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 235, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 31, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 68, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/diagnostics/__init__.py", line 250, in get
    data = await info.config_entry_diagnostics(hass, config_entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 31, in async_get_config_entry_diagnostics
    dumper.create_dump()
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 56, in create_dump
    self._redact_private_information(self.result)
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 129, in _redact_private_information
    self._redact_private_information(value)
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 129, in _redact_private_information
    self._redact_private_information(value)
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 127, in _redact_private_information
    if isinstance(value, str) and _redact_regex.search(key) is not None:
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: expected string or bytes-like object, got 'int'
2024-03-09 16:34:27.573 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 1010, in _update_entity_states
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 963, in async_update_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1113, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1052, in __async_calculate_state
    attr.update(self.state_attributes or {})
                ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/media_player/__init__.py", line 1118, in state_attributes
    if (value := getattr(self, attr)) is not None:
                 ^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/squeezebox/media_player.py", line 323, in media_content_id
    urls = [{"url": track["url"]} for track in self._player.playlist]
                    ~~~~~^^^^^^^
KeyError: 'url'
2024-03-09 16:34:30.316 ERROR (MainThread) [aiohttp.server] Error handling request
[... 141 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 235, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 31, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 68, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/diagnostics/__init__.py", line 250, in get
    data = await info.config_entry_diagnostics(hass, config_entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 31, in async_get_config_entry_diagnostics
    dumper.create_dump()
[... cut ...]
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 129, in _redact_private_information
    self._redact_private_information(value)
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 129, in _redact_private_information
    self._redact_private_information(value)
  File "/config/custom_components/e3dc_rscp/diagnostics.py", line 127, in _redact_private_information
    if isinstance(value, str) and _redact_regex.search(key) is not None:
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: expected string or bytes-like object, got 'int'`

torbennehmer, 2024-03-22:
@kobelka I'll push 3.6.2 shortly, it should contain a fix for the diagnostic dumper handling the error above. Please, next time, paste a log file either as file or at least put it into a code block, so that line breaks are preserved. This took me quite some time to dissect.
Once updated, please attach a diag dump.

kobelka, 2024-04-04:
Sorry for the late reply i was on vacation...
Now the diag dump works: 
[config_entry-e3dc_rscp-613cb8e2575b076366b0dc4c1249b110.json](https://github.com/torbennehmer/hacs-e3dc/files/14864214/config_entry-e3dc_rscp-613cb8e2575b076366b0dc4c1249b110.json)

einmannlan, 2024-07-28:
I wonder if this is related to: s10e_additional_consumption_total?
I have an additional power meter to track my heat pump power draw, but this sensor delivers a value which is negative and growing over time.

RSCPGUI shows the same so it's delivered like this via rscp (I guess).
It would be great to have this value "additional consumption - today"
[image]

Thomansky, 2025-07-19:
It would also be nice if you could read out the kW per phase

TommiG1, 2026-10-23:
This looks already covered by the current integration.

Extra / additional powermeters discovered via RSCP are created as sensors (power and `*-total` energy), including types like additional production and additional consumption. If a meter is missing on your system, a diagnostics dump with `get_powermeters` / `get_powermeters_data` would help tell whether E3DC is not exposing it or something else is wrong.