torbennehmer/hacs-e3dc #121: Invalid authentication

Pendragon1967, 2024-03-09:
### System Health details

I filled all fields.
Username: Your E3DC portal user name
Password: Your E3DC portal password
Hostname: The Hostname or IP address of the E3/DC system
RSCP Password: This is the encryption key used in RSCP communications._
But only responce is: Invalid authentication
what happened? the ip is only xxx.xxx.xxx.xx i think?
thnaks for help

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I filled all fields.
Username: Your E3DC portal user name
Password: Your E3DC portal password
Hostname: The Hostname or IP address of the E3/DC system
RSCP Password: This is the encryption key used in RSCP communications._
But only responce is: Invalid authentication
what happened? the ip is only xxx.xxx.xxx.xx i think?
thnaks for help

### Reproduction steps

I filled all fields.
Username: Your E3DC portal user name
Password: Your E3DC portal password
Hostname: The Hostname or IP address of the E3/DC system
RSCP Password: This is the encryption key used in RSCP communications._
But only responce is: Invalid authentication
what happened? the ip is only xxx.xxx.xxx.xx i think?
thnaks for help

### Debug logs

```text
I filled all fields.
Username: Your E3DC portal user name
Password: Your E3DC portal password
Hostname: The Hostname or IP address of the E3/DC system
RSCP Password: This is the encryption key used in RSCP communications._
But only responce is: Invalid authentication
what happened? the ip is only xxx.xxx.xxx.xx i think?
thnaks for help
```

### Diagnostics dump

I filled all fields.
Username: Your E3DC portal user name
Password: Your E3DC portal password
Hostname: The Hostname or IP address of the E3/DC system
RSCP Password: This is the encryption key used in RSCP communications._
But only responce is: Invalid authentication
what happened? the ip is only xxx.xxx.xxx.xx i think?
thnaks for help

Pendragon1967, 2024-03-09:
Excuse me - im only an user and dont understand this thinks.
i need only help to run this integration.

P37Cloud, 2024-03-12:
same problem here
I checked the controls twice, all is ok, but I cant login

torbennehmer, 2024-03-14:
Please head for the [HA Forum](https://community.home-assistant.io/t/e3dc-remote-storage-control-protocol-rscp/595280/) for this, this is not a bug. Please let us discuss it there, I don't want the issue tracker to be cluttered up with issues like this. Please add the relevant parts from the e3dc integration's debug log always and test against the current version. Also check out the [README](https://github.com/fsantini/python-e3dc) of the pye3dc lib if you've authentication problems, there are a few more notes listed there.

closed 2024-03-14 by torbennehmer