torbennehmer/hacs-e3dc #123: Support for multiple wallboxes

thijsdewitt, 2024-03-19:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I'm not sure if this should be a bug report or an enhancement. 

Configuring 1 wallbox in HA works great. But as soon as I configure the second, the data gets merged. The graphs look identical with a bunch of spikes. So I assume multiple wallboxes are not supported.

**data when 1 Wallbox is configured:**
[image]

**data when 2 Wallboxes are configured:**
[image]

My hardware is 2x S10E_Pro

### Describe the solution you'd like

Support for multiple wallboxes

### Describe alternatives you've considered

Running with only 1 wallbox configured

thijsdewitt, 2024-03-19:
Looking into the code myself, only these values contain incorrect data

https://github.com/torbennehmer/hacs-e3dc/blob/b7686f1f021bb87a1a78c296f694f6df145cba13/custom_components/e3dc_rscp/coordinator.py#L211-L221

torbennehmer, 2024-03-22:
@thijsdewitt You mean you have two independant e3dc unit each with its own IP address?

thijsdewitt, 2024-03-22:
Yes, exactly

torbennehmer, 2024-03-29:
Actually, I have never thought about that possibility, something that's inherently hard to judge for me. I assumed, that one won't have two E3DCs simultaneously.

Honestly, I'm right now very unsure, what happens here. I'd assume, that you have to configure the integration twice, but what happens from there, I don't know.

Do you have any python experience?

thijsdewitt, 2024-03-29:
Still properly testing this, will report back tomorrow when I have solar production data

torbennehmer, 2024-03-30:
@thijsdewitt I'll look into it later, I'm rather busy over Easter. If you want, we can disucss this directly, German or English, as you want. Just contact me directly via torben <at> nehmer <dot> net in this case.

thijsdewitt, 2024-03-30:
No rush, let's discuss this after the holidays.

torbennehmer, 2024-04-10:
@thijsdewitt Thanks for the PR: Out of curiosity, could you give me the diagnostic dumps from your end (I assume that you get two off them, one for each instance of the integration)?

thijsdewitt, 2024-04-10:
[config_entry-e3dc_rscp-5616b5c127cad10ae66d5cab94add41c.json](https://github.com/torbennehmer/hacs-e3dc/files/14936546/config_entry-e3dc_rscp-5616b5c127cad10ae66d5cab94add41c.json)
[config_entry-e3dc_rscp-c4dfb7633b67d92fa0805a41e95290f7.json](https://github.com/torbennehmer/hacs-e3dc/files/14936547/config_entry-e3dc_rscp-c4dfb7633b67d92fa0805a41e95290f7.json)

PR #131 (2024-04-10): Add support for multiple E3DC units 
Fix data fetching when you have multiple e3dc units configured

Fixes #123