torbennehmer/hacs-e3dc #128: SmartGrid Support

tobiby, 2024-03-28:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The S10 mini offers the possibility to control a heat pump via two switches (SmartGrid). It would be nice if it were possible to read out the status of this switches in the Home Assistant, as this would allow you to recognize whether there is a certain amount of excess power and thus switch on consumers

### Describe the solution you'd like

Instead of transferring the status of each individual switch, it might be possible to translate the status of the switch directly into the associated SmartGrid states. 
State 1 - Blocked operation (1:0): Operation for the heat pump is blocked for a maximum of two hours per day.
State 2 - Normal operation (0:0): The heat pump runs in energy-efficient normal operation.
State 3 - Switch-on recommendation (0:1): Heat pump operation is stimulated to increase electricity consumption for heating and hot water.
State 4 - Start-up command (1:1): The heat pump is prompted to operate. This state supports two variants, which must be set on the controller for different tariff and usage models:
i) the heat pump is switched on
ii) the heat pump is switched on AND the hot water temperature is increased

### Describe alternatives you've considered

Provide the state of the switches (0/1) as two separate entities in the system

### Additional context

If [python-e3dc](https://github.com/fsantini/python-e3dc) is used the function poll_switches() can be used to get the state of the switches

torbennehmer, 2024-03-29:
Hi @tobiby,
please add a diagnostic dump.
Torben

tobiby, 2024-05-15:
Created a PR for solving this issue: #147

torbennehmer, 2024-09-08:
Please fix all comments and run the linter over it as well, it should give you hints about incorrect indentations etc.

tobiby, 2024-09-08:
unfortunately i am not familiar with the tools and have no idea how to get the linter running.
If someone could support me I would be happy, if not I would just keep the changes in my fork and cancel the pull request

torbennehmer, 2024-09-10:
in your de container, you should be able to run `python -m ruff check . custom_components/e3dc_rscp/*py` (ignore the config errors it currently throws, that's still on my list).
Please check the other comments in the PR as well. Thanks!

tobiby, 2024-09-11:
[image]

Checks passed. The other comments related to translation i saw and i try to fix like suggested. But i still ask for support because of my lack of knowlegde.

torbennehmer, 2026-02-21:
Released with https://github.com/torbennehmer/hacs-e3dc/releases/tag/v3.11.0