torbennehmer/hacs-e3dc #137: External source or Additional daily total does not match (S10Pro)

Thomansky, 2024-04-10:
### System Health details

The daily total of the external source or the other external source information does not match the values in the system

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The daily total of the external source or the other external source information does not match the values in the system

### Reproduction steps

123

### Debug logs

```text
123
```

### Diagnostics dump

[image]
[home-assistant_e3dc_rscp_2024-04-10T18-25-01.979Z.log](https://github.com/torbennehmer/hacs-e3dc/files/14936616/home-assistant_e3dc_rscp_2024-04-10T18-25-01.979Z.log)

Thomansky, 2024-04-10:
[home-assistant_e3dc_rscp_2024-04-10T18-25-01.979Z.log](https://github.com/torbennehmer/hacs-e3dc/files/14936644/home-assistant_e3dc_rscp_2024-04-10T18-25-01.979Z.log)

torbennehmer, 2024-05-28:
Please add a diagnostic dump and restrict the debug logs to the E3DC component. Also please add the expected values.