torbennehmer/hacs-e3dc #145: Missing sensor for  "Additional Production - today"

Peyto79, 2024-05-02:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

No, I dont think so.

### Describe the solution you'd like

There are several "today" sensors but the one for the additional source is missing. 
"Solar Production - today" does only show the daily sum of the E3DC production and I want to sum it with my additional external production.

### Describe alternatives you've considered

None.

Thomas-sw11, 2024-05-05:
Yes, same for me. Additional today would be nice to have as well.

fummelfichte, 2025-01-12:
+1

I think a combination of helpers (integral followed by utility meter) based on the sensor `S10E_Compact Additional` could be a work-around

TommiG1, 2026-10-23:
Implemented in #409.

That PR adds `Additional production - today` (and also month/year) from the E3DC DB `pm0`/`pm1` totals. Live tested on an S10E Pro with an additional production meter.

closed 2026-10-23 by torbennehmer