torbennehmer/hacs-e3dc #148: hacs-e3dc.git is not a valid add-on repository

knabber, 2024-05-27:
### System Health details

## System Information

version | core-2024.5.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.29-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.05.1
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 30.8 GB
disk_used | 3.8 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | evcc (0.126.4), File editor (5.8.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 26. Mai 2024 um 13:09
-- | --
current_recorder_run | 27. Mai 2024 um 00:22
estimated_db_size | 3.46 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

repository cannot be loaded

### Reproduction steps

1. adding repositry: https://github.com/torbennehmer/hacs-e3dc.git

### Debug logs

```text
2024-05-27 22:35:09.778 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /store/repositories - https://github.com/torbennehmer/hacs-e3dc is not a valid add-on repository
2024-05-27 22:35:21.372 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /store/repositories - https://github.com/torbennehmer/hacs-e3dc.git is not a valid add-on repository
2024-05-27 22:43:34.395 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /store/repositories - https://github.com/torbennehmer/hacs-e3dc.git is not a valid add-on repository
2024-05-27 22:46:15.968 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /store/repositories - https://github.com/torbennehmer/hacs-e3dc.git is not a valid add-on repository
```

torbennehmer, 2024-05-28:
Please enable debug logging and give me the detailed information when you tr to add the repo. Thanks!

torbennehmer, 2024-06-30:
Closed, as we did not get a response. If it still happens, please reopen and provide additional information as requested.

closed 2024-06-30 by torbennehmer