torbennehmer/hacs-e3dc #152: Detected blocking call to open inside the event loop by custom integration e3dc_rscp

Matthias1111, 2024-06-13:
### System Health details

## System Information

version | core-2024.6.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.28-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1386
Downloaded Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 28.0 GB
disk_used | 11.0 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.15.0), Samba share (12.3.1), Network UPS Tools (0.13.0), Advanced SSH & Web Terminal (18.0.0), OpenThread Border Router (2.6.0), Matter Server (6.1.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 2
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 3. Juni 2024 um 07:20
-- | --
current_recorder_run | 13. Juni 2024 um 22:11
estimated_db_size | 282.27 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Integration seems to work fine, but Home Assistant Log shows the following error (and asks to create a bur report):

06/13 22:11:45.033  192.168.0.1  local0.warning (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'e3dc_rscp' at custom_components/e3dc_rscp/coordinator.py, line 267: tz_info: pytz.timezone = pytz.timezone(tz_name) (offender: /usr/local/lib/python3.12/site-packages/pytz/__init__.py, line 108: return open(filename, 'rb')), please create a bug report at https://github.com/torbennehmer/hacs-e3dc/issues
                                                Traceback (most recent call last):
                                                File "<frozen runpy>", line 198, in _run_module_as_main
                                                File "<frozen runpy>", line 88, in _run_code
                                                File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
                                                sys.exit(main())
                                                File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
                                                exit_code = runner.run(runtime_conf)
                                                File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
                                                return loop.run_until_complete(setup_and_run_hass(runtime_config))
                                                File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete
                                                self.run_forever()
                                                File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever
                                                self._run_once()
                                                File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once
                                                handle._run()
                                                File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
                                                self._context.run(self._callback, *self._args)
                                                File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked
                                                await self.async_setup(hass, integration=integration)
                                                File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
                                                result = await component.async_setup_entry(hass, self)
                                                File "/config/custom_components/e3dc_rscp/__init__.py", line 31, in async_setup_entry
                                                await coordinator.async_connect()
                                                File "/config/custom_components/e3dc_rscp/coordinator.py", line 80, in async_connect
                                                await self._load_timezone_settings()
                                                File "/config/custom_components/e3dc_rscp/coordinator.py", line 267, in _load_timezone_settings
                                                tz_info: pytz.timezone = pytz.timezone(tz_name)

### Reproduction steps

1. Restarting HA reproduces the log entries mentioned above

### Debug logs

```text
2024-06-13 22:11:37.234 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'e3dc_rscp' at custom_components/e3dc_rscp/coordinator.py, line 267: tz_info: pytz.timezone = pytz.timezone(tz_name) (offender: /usr/local/lib/python3.12/site-packages/pytz/__init__.py, line 108: return open(filename, 'rb')), please create a bug report at https://github.com/torbennehmer/hacs-e3dc/issues
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked
    await self.async_setup(hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/e3dc_rscp/__init__.py", line 31, in async_setup_entry
    await coordinator.async_connect()
  File "/config/custom_components/e3dc_rscp/coordinator.py", line 80, in async_connect
    await self._load_timezone_settings()
  File "/config/custom_components/e3dc_rscp/coordinator.py", line 267, in _load_timezone_settings
    tz_info: pytz.timezone = pytz.timezone(tz_name)
```

da-anda, 2024-07-03:
I have the same errors in my log. My device is a S10E Blackline

ubarni, 2024-07-15:
Same issue in my logs:

`2024-07-15 11:43:20.919 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/usr/local/lib/python3.12/site-packages/pytz/zoneinfo/Africa/Abidjan', 'rb') inside the event loop by custom integration 'e3dc_rscp' at custom_components/e3dc_rscp/coordinator.py, line 267: tz_info: pytz.timezone = pytz.timezone(tz_name) (offender: /usr/local/lib/python3.12/site-packages/pytz/__init__.py, line 108: return open(filename, 'rb')), please create a bug report at https://github.com/torbennehmer/hacs-e3dc/issues
`

torbennehmer, 2024-07-16:
this will be fixed in the next release

PR #163 (2024-07-16): Fix blocking call to pytz in HA event loop
fixes #152
commit 88ec018d (2024-07-16): Fix blocking call to pytz in HA event loop (#163)

fixes #152
closed 2024-07-16 by torbennehmer