torbennehmer/hacs-e3dc #154: Stricter Linting for smoother PR integration

bullitt186, 2024-06-27:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

When filing a PR, i got to the stage of PR review and only realized then that while working on the PR, i messed up source code formatting. THis means now tedious fixing of the PR. Ideally, I would have been guided earlier in the process that i am deviating from the standard.

### Describe the solution you'd like

Introduce stricter Linting Rules, following the linting strategy + rules the home assistant project itself uses.

Home Assistant uses:
- pylint [source](https://developers.home-assistant.io/docs/development_testing#preventing-linter-errors)
- ruff [source](https://developers.home-assistant.io/docs/development_testing#preventing-linter-errors)
- Black [source] (https://developers.home-assistant.io/docs/development_guidelines#our-recommendations)

THey follow the PEP8 + PEP257 Guidelines [source] (https://developers.home-assistant.io/docs/development_guidelines#our-recommendations)

I suggest to take over the Home Assistant's project configurations to assure adherence to the project's standards and best practices..

In detail this would mean:
- take over the linting aspects of pyproject.toml + it's associated files from [HA core](https://github.com/home-assistant/core/)
- adjust the .devcontainer file accordingly
- add the VSCode extensions for the linters in the .vscode configuration
- Introduce Pre-Commit hooks to ensure compliance to these rules. (See e.g. .pre-commit-config.yaml in the [HA core repo](https://github.com/home-assistant/core/) or [pre-commit.com](https://pre-commit.com/)

### Describe alternatives you've considered

In case strict linting shall not be enforced, the suggested standards could be mentioned in CONTRIBUTING.md

### Additional context

In case you are fine with the suggested approach, i can work on a PR with the suggested changes.

torbennehmer, 2024-07-02:
I think that is a good roadmap. Let's tackle this once we get 3.8 with WB support out of the way. I'd also like to have some rules for the MD files (I've something ready there from work, I'll take that over). Especially, if we (as you rightfully said) add a documentation wiki or something like that.

bullitt186, 2024-07-02:
Sounds good! 
I am happy to work on it (as time allows - as usual :-))

torbennehmer, 2026-10-16:
https://github.com/torbennehmer/hacs-e3dc/releases/tag/v4.0.0

closed 2026-10-16 by torbennehmer