torbennehmer/hacs-e3dc #164: Unfortunately, the service cannot be accessed. Please fix this quickly.  e3dc_rscp.set_power_limits

Thomansky, 2024-07-17:
### System Health details

Unfortunately, the service cannot be accessed. Please fix this quickly. 
e3dc_rscp.set_power_limits

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Leider kann auf den Dienst nicht zugegriffen werden. Bitte beheben Sie das schnell. e3dc_rscp.set_power_limits

### Reproduction steps

1.
2.
3.
H

### Debug logs

```text
.
```

torbennehmer, 2024-07-17:
Incomplete bug report, cannot duplicate

closed 2024-07-17 by torbennehmer

Thomansky, 2024-07-17:
?

torbennehmer, 2024-07-17:
Can't help you without debug logs, diagnostic dump, basically as much information as possible

Thomansky, 2024-07-17:
[config_entry-e3dc_rscp-75379039e826988ca1e6922134bf525b.json](https://github.com/user
[image]
-attachments/files/16269491/config_entry-e3dc_rscp-75379039e826988ca1e6922134bf525b.json)
[image]

[image]

Thomansky, 2024-07-17:
[home-assistant_e3dc_rscp_2024-07-17T16-19-31.281Z.log](https://github.com/user-attachments/files/16269579/home-assistant_e3dc_rscp_2024-07-17T16-19-31.281Z.log)

Thomansky, 2024-07-18:
Need anything else @torbennehmer

torbennehmer, 2024-07-19:
@Thomansky Unfortunately the diag dump link is broken, but from what I deduce: If you're using the current beta, this is probably related to the current changes around the wallbox support in #47, you've been mentioning it there as well IIRC. If so, let's keep it there until we're out of the beta stage and leave this issue closed.
Next time, please upload the logs and the diag dump just in the issue itself, makes things easier.