torbennehmer/hacs-e3dc #175: Action E3DC Remote Storage Control Protocol (Git): Set power limits

thobohoii, 2024-09-02:
### System Health details

## System Information

version | core-2024.8.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.46-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4996
Installed Version | 1.34.0
Stage | running
Available Repositories | 1383
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 30.8 GB
disk_used | 19.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.3.2), Node-RED (18.0.5), evcc (0.130.6), File editor (5.8.0), Terminal & SSH (9.14.0), Mosquitto broker (6.4.1), EMHASS (0.10.6), modbus-proxy (1.0.18), InfluxDB (5.0.0), Grafana (10.0.0), UniFi Network Application (3.2.0), deCONZ (7.0.0), SQLite Web (4.2.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 1
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25. August 2024 um 09:01
-- | --
current_recorder_run | 27. August 2024 um 16:31
estimated_db_size | 910.36 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

If the Maximum charge (w) value is changed via the action E3DC Remote Storage Control Protocol (Git): Set power limits, the Minimum required discharge value is always reset to 100W.

### Reproduction steps

1. change the Minimum required discharge value (Untere Entladeschwelle) via E3DC Portal or RSCPGui for Exempel to 50W
2. take the Home Assistant action E3DC Remote Storage Control Protocol (Git): Set power limits and set withe that the Maximum Charge (Maximale Ladeleistung) value to for expampel 3000W
3. Now take a look back at the Minimum required discharge value (Untere Entladeschwelle) and it will be back at 100W

### Debug logs

```text
Not available
```

bullitt186, 2024-09-04:
I was able to reproduce the issue.
The Service (or now-called action since HA 2024.8) called the underlying pye3dc with "`None`" for the non-set / non-changed values. pye3dc writes the system default value to the limits (`max_charge, max_discharge, min_discharge`) when called with None. Meaning when only changing one of  `max_charge` or `max_discharge`, the other one would be reset to the system value as well in addition to min_discharge. 

To fix this, i changed the service that when the values of one of these three limits are not changed, the currently set value is taken and written (again) via pye3dc to the Hauskraftwerk to preserve the current values.
(The only other option would be the re-implementation of pye3dc's `set_power_limits()` which seemed like overkill to me)

See PR https://github.com/torbennehmer/hacs-e3dc/pull/176

One implication of this bug fix is a change in behaviour of the service. previously, you could (mis)use this service to re-set limits to their default values. now you striclty have to use "clear power limits" to clear all three of them.
However i think this fix makes sense and is more logical than it was before.

@torbennehmer - let's discuss if you think the issue raised by @thobohoii should be handled differently.

thobohoii, 2024-09-04:
@bullitt186 thank you for your investigation. 
Something like that was also my thougth. But i am not the programing guy.

I also tright difrent variations of the Service. Just max_charge or just max_discharge or both.  But always the min_discharge Value will be reset to 100W.

notecp, 2026-04-17:
@torbennehmer what is the state of this? 
@bullitt186 could you make the changes needed for you PR to get included?

or is this already included and i was not able to find it?

torbennehmer, 2026-05-13:
IIRC the system minimum discharge rate only applies if you actually allow for discharching. You can always set the discharge limit to 0, effectively preventing any discharge from the battery.
We'll investige the mentioned PR, besides that, if you control both values simultaneously e.g. by custom integer values, you will be fine.