torbennehmer/hacs-e3dc #185: Use a sensor value in "manual_charge" instead of fixed value

floyd871, 2024-10-02:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I have build an atomation which charges my battery in the night when energy is cheap (Tibber), but i have to put in a fixed value.
I created a sensor which contains the amount of energy i had to put in my battery to charge it to 80%.
But i cannot use that in in the UI-Editor. I put it in the plain-yaml editor, but it don't seem to work.

### Describe the solution you'd like

In the UI choose either a fixed value or a sensor value, like for the "Time Trigger".

### Describe alternatives you've considered

Edit directly in the yaml.

torbennehmer, 2024-10-14:
The basic idea behind the service/action is to give you an entry point to manual charging. I can't tell what sensors you'd need, as this is no reading, but a command to the unit.

closed 2024-10-14 by torbennehmer

floyd871, 2024-10-16:
You completely missunderstood me. I already have a sensor (created myself). I only want the possibility to select it in the GUI instead of entering a fixed value,