torbennehmer/hacs-e3dc #190: Wallbox - Switch between Charging Priority Battery / Wallbox

bs0001, 2024-11-01:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Extension of current functionality

### Describe the solution you'd like

Switching between Wallbox Charging Priority Battery or Wallbox would be quite helpful.

### Describe alternatives you've considered

n/a

mgleisn, 2026-03-09:
I would also love to see this being implemented. 

My usecase would be the automatic charging of the car from the E3DC battery in the morning shortly before leaving to work. Charge Priority would be changed allowing to discharge E3DC for charging the car until a defined threshold is reached (e.g. 30%). Later, when the car is at work and the trehshold would be set back to 100%. This automatism avoids that the E3DC battery is discharged in the afternoon for charging the car. It would allow to use only the energy which was not needed. The goal is to maximize self-consumption and to minimize grid export. This automatism could be even combined with solar forecase prediction (only charge car in the morning if PV forecast that day exceeds a certain threshold).

Matthias1111, 2026-03-18:
I also really would appreciate this feature for HA automations, since this is currently the only one which needs manual interaction. 
Remark: If charging priority is set to Wallbox, there are two new depended options shown (activated for changes) in the E3 DC app (see screenshot in [discussion 228](https://github.com/torbennehmer/hacs-e3dc/discussions/228)).

torbennehmer, 2026-10-16:
Implemented in https://github.com/torbennehmer/hacs-e3dc/releases/tag/v4.0.0

closed 2026-10-16 by torbennehmer