torbennehmer/hacs-e3dc #199: Connect to E3DC Farm

Hans-Gg, 2024-11-08:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

When using several E3DC home power stations in farming mode, one device is the master and the connection must be made on port 5034 instead of 5033.

### Describe the solution you'd like

Could this be taken into account when configuring the device?

### Describe alternatives you've considered

No alternatives known

torbennehmer, 2024-11-23:
Is it just a different port or do we have additional requirements here? 
Unfortunately, this requires a change in pye3dc, the connection port isn't exposed by the library at the moment. Would you mind creating an issue [at the pye3dc project](https://github.com/fsantini/python-e3dc/), you can reference both me and this issue. I can't promise when fsantini will get around to this or when/if I've the time to create a PR for pye3dc, but the fix has to start there in any case.

For my own archive: Would it be possible to get a diagnostic dump for all involved units at some point

Hans-Gg, 2024-11-23:
Yes, it´s just a different port. There are no additional requirements.
I will creating an issue at the pye3dc project.
Thank you very much.

volkerrichert, 2025-12-29:
Great to hear. Can you describe your setup and what options do you use from the e3dc system? 

The remote IP isn't an IP or valid hostname. Are you using AI360 oder SG or some other feature which hast to be enabled?