torbennehmer/hacs-e3dc #202: Ability to get battery SOH

volkerrichert, 2024-11-16:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Providing of S10 battery SOH would be quite helpful.

### Describe the solution you'd like

n/a

### Describe alternatives you've considered

n/a

jh537, 2024-11-17:
Describe alternatives you've considered: Sensor in HA

volkerrichert, 2024-11-17:
I changed the request

torbennehmer, 2024-11-17:
See #204, there are a few thoughts, I'd like your feedback on.

jensihnow, 2026-01-03:
Just to close the loop, I just installed the integration using the beta https://github.com/torbennehmer/hacs-e3dc/releases/tag/v3.11.0-beta.2 and when enabled the battery packs and batteries are showing up with plenty of sensors. 

I did noticed a few sensors:

- Parallel cell count
- Series cell count

aren't showing up with an icon, please see:

[image]

I didn't had the time to take a closer look.

jensihnow, 2026-01-03:
Looks like 

> mdi:battery-multiple

 doesn't exist... see e.g., https://github.com/torbennehmer/hacs-e3dc/blob/db1eea69ff12199ffdb314b7697ea4493406957a/custom_components/e3dc_rscp/sensor.py#L520 

@bullitt186 you might want to take a look and fix the icon for both of the sensors.