torbennehmer/hacs-e3dc #203: Display of the battery charge cycles as a sensor

jh537, 2024-11-17:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The value is stored in the S10 and it is important to monitor it. Since I use the integration, it should also be timed with it. Otherwise I would have to install something else, e.g. IO-Broker. 

### Describe the solution you'd like

I imagine a sensor in HA that I can read out and use for further calculations.

### Describe alternatives you've considered

I would have to install something else, e.g. IO-Broker.

torbennehmer, 2024-11-17:
See #204, there are a few thoughts, I'd like your feedback on.

torbennehmer, 2026-05-17:
This has been fixed in 3.11.0, the new (optional) battery pack devices show these values.