torbennehmer/hacs-e3dc #208: GET/SET Price limit for charging the E3DC battery

ubarni, 2024-12-13:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

In the new E3DC portal it is now possible to set the price limit for charging the E3DC battery.
This would be a great feature to get and set this value in Home Assistant to automate the charging of the E3DC battery.
In HA this will be more flexible (i.e. using automations).

### Describe the solution you'd like

Enhance e3dc_rscp with a new method to set the price limit for charging the E3DC battery.

e3dc_rscp.set_charge_price_limit
data:
  device_id: yourDeviceId
  charge_price_limit: 0.000 - 999.999 (ct/kWh)

Add a new sensor "charge_price_limit" to show the actual price limit for charging the E3DC battery.

### Describe alternatives you've considered

Meanwhile it is possible to set the price limit for charging the E3DC battery in the E3DC portal.

ubarni, 2024-12-13:
[image]

bullitt186, 2025-12-03:
I opened a [PR at python-e3dc](https://github.com/fsantini/python-e3dc/pull/138) which adds the features in the python lib this HA integration builds on.
Once the PR is accepted/merged, we can look into adding this feature also to the HA integration