torbennehmer/hacs-e3dc #209: Home Assistent 2024.12

Adi146, 2024-12-14:
### System Health details

## System Information

version | core-2024.12.3
-- | --
installation_type | Unknown
dev | false
hassio | false
docker | false
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 5.15.0-125-generic
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1489
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 1
views | 20
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 4. Dezember 2024 um 22:04
-- | --
current_recorder_run | 13. Dezember 2024 um 18:56
estimated_db_size | 959.06 MiB
database_engine | postgresql
database_version | 16.6

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hacs-e3dc stopped Wirkung after the update to Home Assistent 2024.12.0 in Containers. It looks Luke the underlying Python lib is not built with Python 3.13 which was introducted with thin HA Version. 

### Reproduction steps

1. Start Home Assistent in docker with 2024.12.0 or above.
2. Configure hacs-e3dc

### Debug logs

```text
Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 13. Dezember 2024 um 18:56:23 (1 Vorkommnisse)
Zuletzt protokolliert: 13. Dezember 2024 um 18:56:23

Setup failed for custom integration 'e3dc_rscp': Unable to import component: No module named 'e3dc'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1034, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1014, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/e3dc_rscp/__init__.py", line 22, in <module>
    from .coordinator import E3DCCoordinator
  File "/config/custom_components/e3dc_rscp/coordinator.py", line 10, in <module>
    from e3dc._rscpTags import PowermeterType
ModuleNotFoundError: No module named 'e3dc'
```

torbennehmer, 2024-12-18:
I cannot reproduce this when bootstrapping everything in 3.13 python. My Dev Container works as usual with 2024.12 and Py 3.13. 

pye3dc itself is configured to require python 3.8 or greater, there is no upper limit.

I've just bumped a dev branch to require HA 2024.12 and will publish a beta release. If it still doesn't work, I'd recommend deploying a new container and looking at the entire startup. At some point the pye3dc libs should be loaded, the root cause is probably visible there. Without further information, I have no idea beyond formally updating hacs-e3dc's dependencies

torbennehmer, 2024-12-18:
https://github.com/torbennehmer/hacs-e3dc/releases/tag/v3.9.0-beta.1 available, please validate against this release. If the problem persists, provide additional information as outlined above and reopen.

Adi146, 2024-12-27:
I found the related issue in the ha repo https://github.com/home-assistant/core/issues/133537

closed 2025-01-10 by torbennehmer