torbennehmer/hacs-e3dc #215: Use TAG_EMS_REQ_SET_POWER for battery charging

torbennehmer, 2025-01-15:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/torbennehmer/hacs-e3dc/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The current battery charging using manual charge is suboptimal, as E3DC limits its availability. It should be replaced by the ioBroker approach, using TAG_EMS_REQ_SET_POWER

### Describe the solution you'd like

Check the functionality of TAG_EMS_REQ_SET_POWER and implement it. It needs to be discussed, what options are sensible here, once the interface is understood.

See also sendEmsSetPower in ioBroker:
https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/main.js#L923

Alternative permalink, probably out of date
https://github.com/git-kick/ioBroker.e3dc-rscp/blob/a1e3179b1578d7c9976eabd0bd1eca0992d5207d/main.js#L923-L951

### Describe alternatives you've considered

Using EMS Manual Charging is instable and difficult to judge, so it basically has to be ruled out.

### Additional context

Split of #211

torbennehmer, 2025-01-15:
@volkerrichert

If you want to give it a shot, adding it to HA is only the second step. Try to make it work on the command line, with a simple Python PoC using pye3dc directly. I don't completely understand yet, what ioBroker's doing there, and what opportunities it gives beyond simple charging. If there's a way to regulate what the unit is doing, you could even have more dynamics in there (for whatever that could be used on the long run).

Moving it into the integration would be the second step then.

@bullitt186 FYI, this might be of interest to you too

volkerrichert, 2025-01-17:
ok, i'll take a look at pye3dc first

volkerrichert, 2025-01-17:
pye3dc defines the TAGs, but doesn't implement it. It's obvious why.... e3dc will reset this state after a few seconds (AFAIK) and it has to be re-set after 10 (or less) seconds

volkerrichert, 2025-01-17:
By default iobroker-e3dc does this every 10 sec. But I saw some charges being ended without any interactions after a 1-2 hours. I'm testing 8sec

torbennehmer, 2025-02-02:
The official E3DC tag documentation (you can download it from the old portal, don't know where it is on the new one) formally defines that any non-standard mode set by this command will be reset after 30 seconds, which is why ioBroker updates it continously.

volkerrichert, 2025-02-28:
@torbennehmer I had time to start implementing and make some basic steps in coding HA. But the pyE3DC doesn't any of those requests. But the lib doesn't seems to be actively maintained. Even basic requests are answered.

What do you think about polling & setting data using e3dc sendRequest directly in the integration and bypass the provided methods? 

Volker

torbennehmer, 2025-03-02:
@volkerrichert Yes, you can do that. Please look at the e3dc proxy, all code related to e3dc communication should be in there. You'll find a number of direct sendRequests there as well. The coordinator, sensors etc. must only use the e3dc proxy (the idea is to move that part to the pye3dc (or a fork of it) at some point).
The diagnostics code is an exception, as I use it to collect statistics about E3DC units. Its bypassing the proxy in a few places to collect statistics the integration doesn't support at this time. So don't use that part of the code as an example.

volkerrichert, 2025-03-02:
@torbennehmer I implemented most of the basic sensors, selects and numbers for learning purpose. But at this point, I'm not sure if this should be a service instead of controls. Both SET_POWER_MODE and SET_POWER_VALUE just needed to control the battery mode which feels like parameters to a service.

But mode just reflects "CHARGING" and hide "GRID_CHARGING".

Any hint about the "HA" way doing it?

volkerrichert, 2025-06-20:
Hi Torben,

I created a PR a few days ago solving that issue.

https://github.com/torbennehmer/hacs-e3dc/pull/240